arduino-libraries/MKRNB #12: cannot get SSL connections to work

eikaramba, 2019-04-21:
Is there any documentation about connecting to SSL websites?
If i try the example from https://www.arduino.cc/en/Tutorial/MKRNBExamplesNBSSLWebClient it does not send the request. It works with http, but i tried several different ssl requests and ssl never works.

00:40:03.016 -> connecting...
00:40:23.535 -> connected

sandeepmistry, 2019-04-23:
Hi @eikaramba,

The example is working for me, what version of the library are you using?

Have you tried any other HTTPS servers? The root certs the library is bundled with can be found here: https://github.com/arduino-libraries/MKRNB/blob/master/src/utility/NBRootCerts.h

eikaramba, 2019-04-28:
I tried multiple different ssl domains. currently testing with https://webhook.site, and again it works when i'm using port 80 and NBClient(you can call that site with or without ssl), but not if i'm switching to 443 and NBSSLClient.

The behaviour is always like this:
```
Serial.println("connected");
    // Make a HTTP request:
    client.print("GET ");
    Serial.println("test");
    client.print(path);
    Serial.println("test2");
    client.println(" HTTP/1.1");
    client.print("Host: ");
    client.println(server);
    Serial.println("test3");
    client.println("Connection: close");
    client.println();
    Serial.println("finally");
```

```
13:57:26.656 -> Starting Arduino web client.
13:57:44.083 -> connecting...
13:58:03.798 -> connected
13:58:13.867 -> test
```

So you see it is basically not doing anything anymore after the GET command. 

If i do the SAME exact code and now just change port to 80 and use the non ssl lib, it works.

```
14:01:30.686 -> Starting Arduino web client.
14:01:45.113 -> connecting...
14:01:49.533 -> connected
14:01:49.580 -> test
14:01:49.627 -> test2
14:01:49.768 -> test3
14:01:49.861 -> finally
14:02:00.116 -> HTTP/1.1 200 OK
14:02:00.116 -> Server: nginx/1.10.3
14:02:00.116 -> Content-Type: text/plain; charset=UTF-8
14:02:00.350 -> Transfer-Encoding: chunked
14:02:00.819 -> Connection: close
14:02:00.819 -> Vary: Accept-Encoding
14:02:02.053 -> X-Request-Id: 989ce15e-549b-4614-8f71-c1ff4524f591
14:02:04.021 -> X-Token-Id: 592ad91d-b3a2-4d6b-9b3f-0733a8159089
14:02:04.256 -> Cache-Control: no-cache, private
14:02:04.256 -> Date: Sun, 28 Apr 2019 12:01:57 GMT
14:02:04.505 -> X-RateLimit-Limit: 30
14:02:04.633 -> X-RateLimit-Remaining: 29
14:02:04.633 -> 
14:02:04.633 -> 0
14:02:04.633 ->
```

Also it seems to be much faster. Do you have any idea what is going on here? I even tried to use the RAW AT commands inspired by this script here (https://github.com/telstra/arduino-mkr-nb-1500-cumulocity/blob/master/send_measurements/send_measurements.ino).  Not working as well unfortunately.

I'm using the latest versions of everything as of today. Really glad for every tip you can give.

sandeepmistry, 2019-05-01:
If you change the `NB nbAccess;` of the sketch to `NB nbAccess(true);` we can get some more debug input.

What I suspect is happening is the SSL connection is failing and `.connect(...)` returns 2 instead of 1, this is a port over from the original GSM library. However, this seems not inline with the `Client` API spec.

@cmaglie @facchinm I'll prepare a pull request to change the return value to 0 on failure.

sandeepmistry, 2019-05-01:
Please ignore my previous comment, as it only applies to the MKR GSM lib.

@eikaramba if you can provide some debug logs as I indicated above that would be great.

sandeepmistry, 2019-05-03:
@eikaramba I've tried the following sketch on two of my boards, and all looks fine:

```
/*
  SSL Web client

 This sketch connects to a website using SSL through a MKR NB 1500 board. Specifically,
 this example downloads the URL "https://www.arduino.cc/asciilogo.txt" and
 prints it to the Serial monitor.

 Circuit:
 * MKR NB 1500 board
 * Antenna
 * SIM card with a data plan

 created 8 Mar 2012
 by Tom Igoe
*/

// libraries
#include <MKRNB.h>

#include "arduino_secrets.h" 
// Please enter your sensitive data in the Secret tab or arduino_secrets.h
// PIN Number
const char PINNUMBER[]     = SECRET_PINNUMBER;

// initialize the library instance
NBSSLClient client;
GPRS gprs;
NB nbAccess;

// URL, path and port (for example: arduino.cc)
char server[] = "webhook.site";
char path[] = "/";
int port = 443; // port 443 is the default for HTTPS

void setup() {
  // initialize serial communications and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  Serial.println("Starting Arduino web client.");
  // connection state
  boolean connected = false;

  // After starting the modem with NB.begin()
  // attach to the GPRS network with the APN, login and password
  while (!connected) {
    if ((nbAccess.begin(PINNUMBER) == NB_READY) &&
        (gprs.attachGPRS() == GPRS_READY)) {
      connected = true;
    } else {
      Serial.println("Not connected");
      delay(1000);
    }
  }

  Serial.println("connecting...");

  // if you get a connection, report back via serial:
  if (client.connect(server, port)) {
    Serial.println("connected");
    // Make a HTTP request:
    client.print("GET ");
    client.print(path);
    client.println(" HTTP/1.1");
    client.print("Host: ");
[... 797 lines cut ...]
    </div><!-- /.modal -->

    <script type="application/javascript">
        window.ga = window.ga || function () {
            (ga.q = ga.q || []).push(arguments)
        };
        ga.l = +new Date;
        ga('create', 'UA-5230636-9', 'auto');
        ga('require', 'eventTracker');
        ga('require', 'outboundLinkTracker');
        ga('require', 'urlChangeTracker');
        ga('require', 'pageVisibilityTracker');
        ga('send', 'pageview');
    </script>
    <script async src="https://www.google-analytics.com/analytics.js"></script>
    <script async src="assets/scripts/libs/autotrack.js"></script>
</div>
</body>
</html>

0

disconnecting.

```

eikaramba, 2019-05-06:
thanks for your help.  i activated the debug mode and got the following. seems there is a problem with connecting to the server via socket?
```
23:26:18.496 -> Starting Arduino web client.
23:26:18.599 -> AT

23:26:18.599 -> OK
23:26:18.734 -> AT+CFUN=0

23:26:18.734 -> OK
23:26:18.940 -> AT+CGATT=0

23:26:18.940 -> OK
23:26:19.143 -> AT+CPIN?

23:26:19.143 -> +CPIN: READY
23:26:19.143 -> 
23:26:19.143 -> OK
23:26:19.347 -> AT+CMGF=1

23:26:19.347 -> OK
23:26:19.550 -> AT+UDCONF=1,1

23:26:19.550 -> OK
23:26:19.753 -> AT+CTZU=1

23:26:19.753 -> OK
23:26:19.955 -> AT+CGDCONT=1,"IP",""

23:26:19.955 -> OK
23:26:20.158 -> AT+CFUN=1

23:26:20.158 -> OK
23:26:20.362 -> AT+CEREG?

23:26:20.362 -> +CEREG: 0,0
23:26:20.362 -> 
23:26:20.362 -> OK
23:26:20.566 -> AT+CEREG?

23:26:20.566 -> +CEREG: 0,0
23:26:20.566 -> 
23:26:20.566 -> OK
23:26:20.766 -> AT+CEREG?

23:26:20.766 -> +CEREG: 0,0
23:26:20.766 -> 
23:26:20.766 -> OK
23:26:20.969 -> AT+CEREG?

23:26:20.969 -> +CEREG: 0,0
23:26:20.969 -> 
23:26:20.969 -> OK
23:26:21.170 -> AT+CEREG?

23:26:21.170 -> +CEREG: 0,0
23:26:21.170 -> 
23:26:21.170 -> OK
23:26:21.373 -> AT+CEREG?

23:26:21.373 -> +CEREG: 0,0
23:26:21.373 -> 
23:26:21.373 -> OK
23:26:21.576 -> AT+CEREG?

23:26:21.576 -> +CEREG: 0,0
23:26:21.576 -> 
23:26:21.576 -> OK
23:26:21.775 -> AT+CEREG?

23:26:21.775 -> +CEREG: 0,0
[... 830 lines cut ...]
23:29:26.186 -> 
23:29:26.186 -> +UUSORD: 0,48
23:29:26.186 -> AT+USORD=0,512

23:29:26.186 -> 
23:29:26.186 -> +USORD: 0,48,"6E743D226E6F696E6465782C6E6F666F6C6C6F7722202F3E0A20202020202020203C7374796C653E2020202020202020"
23:29:26.233 -> OK
23:29:26.233 -> nt="noindex,nofollow" />
23:29:26.233 ->         <style>        
23:29:26.233 -> +UUSORD: 0,48
23:29:26.233 -> AT+USORD=0,512

23:29:26.233 -> 
23:29:26.233 -> +USORD: 0,48,"2020202020202020626F6479207B206261636B67726F756E642D636F6C6F723A20236666663B20636F6C6F723A202332"
23:29:26.233 -> OK
23:29:26.233 ->         body { background-color: #fff; color: #2AT+USORD=0,512

23:29:26.280 -> +USORD: 0,""
23:29:26.280 -> 
23:29:26.280 -> OK
23:29:26.280 -> AT+USORD=0,512

23:29:26.280 -> +USORD: 0,""
.... and so on
```

sandeepmistry, 2019-05-13:
This log looks perfectly ok, to me:

```
23:29:23.819 -> AT+USOCO=0,"webhook.site",80

23:29:24.438 -> OK
23:29:24.438 -> connected
```
^ indicates a successful connection

```
23:29:25.597 -> +USORD: 0,48,"485454502F312E3120343034204E6F7420466F756E640D0A5365727665723A206E67696E782F312E31302E330D0A436F"
23:29:25.597 -> OK
23:29:25.597 -> HTTP/1.1 404 Not Found
23:29:25.597 -> Server: nginx/1.10.3
23:29:25.597 -> CoAT+USORD=0,512

....
```
^ indicates a HTTP response from the server.

eikaramba, 2019-05-13:
@sandeepmistry please have a look again. i posted 2 logs, one with ssl and one without. the one that worked was without ssl.
in the one with ssl you see:
```
23:26:48.525 -> +CME ERROR: Operation not allowed
23:26:48.559 -> 
23:26:48.559 -> AT+USOCL=0
test
23:27:01.493 -> 
23:27:01.493 -> OK
```
at the end

sandeepmistry, 2019-05-14:
Hi @eikaramba,

Sorry, I missed it ...

A few questions for you:

1) Just to confirm you are using the [NBSSLWebClient.ino](https://github.com/arduino-libraries/MKRNB/blob/master/examples/NBSSLWebClient/NBSSLWebClient.ino) example with the server changed to "webhook.site" ?

2) Does the example work as is (with the arduino.cc domain)?

eikaramba, 2019-05-14:
Yes exactly. It's the exact same code apart from the different domain.
Using arduino.cc domain i have the same problem. it works withhout ssl, but not with ssl.

sandeepmistry, 2019-05-14:
Hmm, so it might be either of:

1) the carrier is disabled SSL.
2) the u-blox module can't sync the current time with the cellular network.

Some suggestions:
a) try another SIM if possible
b) add a `Serial.println(nbAccess.getTime());` after the failed connection. To print the current time of the module.

eikaramba, 2019-12-23:
i finally had more time to debug this, but unfortunately it is still not working.

1) i tried to print the local time and got the following:
```
OK
Starting modem test...AT
OK
modem.begin() succeeded
Local Time-----------
AT+CCLK?
+CCLK: "80/01/06,00:00:16"

OK
315964816
Time-----------
AT+CCLK?
+CCLK: "80/01/06,00:00:16"
```

this gives me the time since the device powered on(here 16 seconds). Is that right? I don't think so. but how can i "reset" or "set" the time?

the example with the ntp udp packet to get the time resulted in that:

```
Starting connection to server...
packet received
Seconds since Jan 1 1900 = 3786109524
Unix time = 1577120724
The UTC time is 17:05:24
packet received
Seconds since Jan 1 1900 = 3786109535
Unix time = 1577120735
The UTC time is 17:05:35
```

2) i activated the debug via `NB nbAccess(true);` and once more tried to call webhook.site on port 443  via the nbsslclient example:

```
Starting Arduino web client.
AT

OK
AT

OK
AT+CMEE=0

OK
AT+CFUN=0

OK
AT+CPIN?

+CPIN: READY

OK
AT+CMGF=1

OK
AT+UDCONF=1,1

OK
AT+CTZU=1

OK
AT+CGDCONT=1,"IP",""

OK
AT+UAUTHREQ=1,0

OK
AT+CFUN=1

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,5

OK
AT+CGATT=1

OK
AT+CGACT?

+CGACT: 1,1

OK
connecting...
AT+USECMNG=0,0,"AddTrust_External_CA_Root",1082

>
+USECMNG: 0,0,"AddTrust_External_CA_Root","1D3554048578B03F42424DBF20730A3F"

OK
AT+USECMNG=0,0,"Baltimore_CyberTrust_Root",891

>
+USECMNG: 0,0,"Baltimore_CyberTrust_Root","ACB694A59C17E0D791529BB19706A6E4"

OK
AT+USECMNG=0,0,"COMODO_RSA_Certification_Authority",1500

>
+USECMNG: 0,0,"COMODO_RSA_Certification_Authority","1B31B0714036CC143691ADC43EFDEC18"

OK
AT+USECMNG=0,0,"DST_Root_CA_X3",846

>
+USECMNG: 0,0,"DST_Root_CA_X3","410352DC0FF7501B16F0028EBA6F45C5"

OK
AT+USECMNG=0,0,"DigiCert_High_Assurance_EV_Root_CA",969

>
+USECMNG: 0,0,"DigiCert_High_Assurance_EV_Root_CA","D474DE575C39B2D39C8583C5C065498A"

OK
AT+USECMNG=0,0,"Entrust_Root_Certification_Authority",1173

>
+USECMNG: 0,0,"Entrust_Root_Certification_Authority","D6A5C3ED5DDD3E00C13D87921F1D3FE4"

OK
AT+USECMNG=0,0,"Equifax_Secure_Certificate_Authority",804

>
+USECMNG: 0,0,"Equifax_Secure_Certificate_Authority","67CB9DC013248A829BB2171ED11BECD4"

OK
AT+USECMNG=0,0,"GeoTrust_Global_CA",856

>
+USECMNG: 0,0,"GeoTrust_Global_CA","F775AB29FB514EB7775EFF053C998EF5"

OK
AT+USECMNG=0,0,"GeoTrust_Primary_Certification_Authority_G3",1026

>
+USECMNG: 0,0,"GeoTrust_Primary_Certification_Authority_G3","B5E83436C910445848706D2E83D4B805"

OK
AT+USECMNG=0,0,"GlobalSign",958

>
+USECMNG: 0,0,"GlobalSign","9414777E3E5EFD8F30BD41B0CFE7D030"

OK
AT+USECMNG=0,0,"Go_Daddy_Root_Certificate_Authority_G2",969

>
+USECMNG: 0,0,"Go_Daddy_Root_Certificate_Authority_G2","803ABC22C1E6FB8D9B3B274A321B9A01"

OK
AT+USECMNG=0,0,"VeriSign_Class_3_Public_Primary_Certification_Authority_G5",1239

>
+USECMNG: 0,0,"VeriSign_Class_3_Public_Primary_Certification_Authority_G5","CB17E431673EE209FE455793F30AFA1C"

OK
AT+USECMNG=2,0,"AmazonRootCA1"

ERROR
AT+USECMNG=0,0,"Starfield_Services_Root_Certificate_Authority_G2",1011

>
+USECMNG: 0,0,"Starfield_Services_Root_Certificate_Authority_G2","173574AF7B611CEBF4F93CE2EE40F9A2"

OK
AT+USOCR=6

+USOCR: 0

OK
AT+USOSEC=0,1,0

OK
AT+USECPRF=0,0,1

OK
AT+USOCO=0,"webhook.site",443

OK
connected
AT+USOWR=0,4,"47455420"

ERROR
AT+USOWR=0,37,"2F65663435656532642D613331322D343632352D623539382D306262343233303161373161"

ERROR
AT+USOWR=0,9,"20485454502F312E31"

ERROR
AT+USOWR=0,2,"0D0A"

ERROR
AT+USOWR=0,6,"486F73743A20"

ERROR
AT+USOWR=0,12,"776562686F6F6B2E73697465"

ERROR
AT+USOWR=0,2,"0D0A"

ERROR
AT+USOWR=0,17,"436F6E6E656374696F6E3A20636C6F7365"

ERROR
AT+USOWR=0,2,"0D0A"

ERROR
AT+USOWR=0,2,"0D0A"

ERROR
AT+USORD=0,512

+USORD: 0,""

OK
AT+USORD=0,512

+USORD: 0,""

OK
AT+USORD=0,512

+USORD: 0,""

........
.......
......
hundred more of those USORD entries
......

+UUSOCL: 0
AT+USORD=0,512

ERROR
AT+USOCL=0

ERROR

disconnecting.

```

eikaramba, 2019-12-23:
i tried to manually send the http call via the http ubox module https://www.u-blox.com/sites/default/files/SARA-R4_ATCommands_%28UBX-17003787%29.pdf#%5B%7B%22num%22%3A694%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C59.527%2C742.677%2Cnull%5D
```
AT+UHTTP=0,1,"webhook.site"
AT+UHTTP=0,5,443
AT+UHTTP=0,6,1,0
AT+UHTTP=0,4,0
AT+UHTTPC=0,1,"/ef45ee2d-a312-4625-b598-0bb42301a71a","res"
OK
```
However this gives the following http error state
```
+UUHTTPCR: 0,1,0

AT+UHTTPER=0
+UHTTPER: 0,3,13
```
Looking at the manual this indicated an "internal error" in the HTTP Protocol 
https://www.u-blox.com/sites/default/files/SARA-R4_ATCommands_%28UBX-17003787%29.pdf#%5B%7B%22num%22%3A865%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C59.527%2C584.377%2Cnull%5D

crcer, 2020-10-14:
Do you have any solution for this yet?
I am struggling with the same issue using a SimCard from 1nce.

eikaramba, 2020-10-14:
unfortunately i have given up. i really tried everything even tried to call all the AT commands manually myself etc. but i never got SSL working. at the end i create a http proxy on my server which basically handles the ssl handshake.

dkoffler, 2020-11-28:
First of all, it looks like you are not getting the proper time from the cell network! It's outputting a date in 1980. This is probably messing a whole bunch of things up.

There are 2 things I would try...

1) Upgrade the Ublox modem firmware. Once I did that, a whole bunch of problems I was having went away. This is a good video on how to do that: https://www.youtube.com/watch?v=33LCz9fjyes Just note that there are 2 different downloads for the firmware update files depending on which version of firmware you currently have [A0200](https://ublox-firmware.s3.amazonaws.com/L0506A0200-to-L0508A0204.zip) or [A0201](https://ublox-firmware.s3.amazonaws.com/L0506A0201-to-L0508A0204.zip). You can see which version you have after you install the Ublox tool described in the video.

2) If that doesn't help, then try a different SIM from a different provider as previously suggested. the fact that you are not getting the correct time indicates the modem is having trouble communicating with the network. Are you sure you have CAT M1 or  NB-IoT support in your area?

netskink, 2021-09-04:
Any method for adding  a new root cert?  I tried getting the cert with openssl s_client connect command.  I took text between begin/end signature, decoded base64 and then put resulting hex and length in the header file.  Sadly when it comes time to load the cert it fails.

Hugocourr, 2022-04-27:
> Any method for adding a new root cert? I tried getting the cert with openssl s_client connect command. I took text between begin/end signature, decoded base64 and then put resulting hex and length in the header file. Sadly when it comes time to load the cert it fails.

For updating new SSl certificate i made a tutorial on the MKRGSM that is easely portable onto the MKRNB (you just need to change the type of client to the one you usually use, here is the topic (see 2nd comment for the tutorial) https://github.com/arduino-libraries/MKRGSM/issues/151
Hope it will help you!