arduino-libraries/MKRNB #20: +CME ERROR: Operation not allowed - randomkly

guydillen, 2019-06-03:


Rocketct, 2019-06-05:
Hi @guydillen could you share the log of the at command?

guydillen, 2019-06-05:
Hi @Rocketct , in the mean time I already changed y sketch a little bit by adding a separate connect function, but obviously I still get this CME ERROR message after a while.

This is the serial output:

```
23:19:35.788 -> Attempting to connect to the cellular network
23:19:40.115 -> AT

23:19:40.115 -> OK
23:19:40.218 -> AT

23:19:40.218 -> OK
23:19:40.218 -> AT+CMEE=0

23:19:40.218 -> OK
23:19:40.424 -> AT+CFUN=0

23:19:40.424 -> OK
23:19:40.630 -> AT+CPIN?

23:19:40.630 -> +CPIN: READY
23:19:40.630 -> 
23:19:40.630 -> OK
23:19:40.838 -> AT+CMGF=1

23:19:40.838 -> OK
23:19:41.044 -> AT+UDCONF=1,1

23:19:41.044 -> OK
23:19:41.215 -> AT+CTZU=1

23:19:41.215 -> OK
23:19:41.422 -> AT+CGDCONT=1,"IP",""

23:19:41.422 -> OK
23:19:41.630 -> AT+UAUTHREQ=1,0

23:19:41.630 -> OK
23:19:41.836 -> AT+CFUN=1

23:19:41.836 -> OK
23:19:42.041 -> AT+CEREG?

23:19:42.041 -> +CEREG: 0,0
23:19:42.041 -> 
23:19:42.041 -> OK
23:19:42.249 -> AT+CEREG?

23:19:42.249 -> +CEREG: 0,0
23:19:42.249 -> 
23:19:42.249 -> OK
23:19:42.456 -> AT+CEREG?

23:19:42.456 -> +CEREG: 0,0
23:19:42.456 -> 
23:19:42.456 -> OK
23:19:42.662 -> AT+CEREG?

23:19:42.662 -> +CEREG: 0,0
23:19:42.662 -> 
23:19:42.662 -> OK
23:19:42.834 -> AT+CEREG?

23:19:42.869 -> +CEREG: 0,0
23:19:42.869 -> 
23:19:42.869 -> OK
23:19:43.041 -> AT+CEREG?

23:19:43.041 -> +CEREG: 0,0
23:19:43.041 -> 
[... 601 lines cut ...]
  if (nbAccess.status() != NB_READY || gprs.status() != GPRS_READY) {
    connectNB();
  }
  sendUDPMessage(awsServer);
  delay(900000);
}

void connectNB() {
  Serial.println("Attempting to connect to the cellular network");

  while ((nbAccess.begin() != NB_READY) ||
         (gprs.attachGPRS() != GPRS_READY)) {
    // failed, retry
    Serial.print(".");
    delay(1000);
 }
 
  Serial.println("You're connected to the cellular network");
  Serial.println();

  Serial.println("\nStarting connection to server...");
  // Initialize, start listening on specified port
  udp.begin(localPort);
}
```

lekron42, 2020-02-14:
I have the same problem.. It just randomly fails to create a socket. 

I noticed following behaviour:
Uploading sketch --> Everything works
Unplugging it and connecting it again --> CME Error: Operation not allowed

Since my device should run without a serial monitor connected, i use a watchdog to restart the system whenever this error occurs but this can result in more than 5 loops until it is able to create the sockets..

CptHolzschnauz, 2021-07-23:
I had similar flaky problems. The cause was too many AT commands in too little time. The modem needs time to digest the commands.

furqon2710, 2023-11-30:
so is there any way to catch this failure by the library and do restart by it self when having this problem?

SalomehW, 2025-03-06:
I had the same problem, and catching the error and restarting the modem worked for me. 

String listenCommand = "AT+USOLI=" + String(socketID) + ",2323";
  String response = sendATCommand(listenCommand.c_str());

  if (response.indexOf("+CME ERROR: Operation not allowed") != -1) {
    Serial.println("[ERROR] USOLI failed. Try again...");
    nbAccess.shutdown();
    delay(10000);       
    setup();
  }
  Serial.println("[INFO] TCP-Server läuft auf Port 23...");
}