arduino-libraries/MKRNB #22: MKR NB1500 - mqttClient.connect(broker, port) retry failed

splitout, 2019-06-23:
Hi all, 

Iam actually testing the project: 
Arduino_Cloud_Provider_Examples -> Azure IoT Hub -> Azure_IoT_Hub_NB

In the most cases the "mqttClient.connectError()" returns the value "3". 
Error-Code for: "MQTT_SERVER_UNAVAILABLE "

Out 1 of 100 tests to connect the mqttClient works fine. Then the connection to the azure-hub is successfully and I can do what I want to send datas to the Azure-Hub.
But if I start the sketch again, the mqttclint.connect() shows an error (MQTT_SERVER_UNAVAILABLE) again and then the NB 1500 crashed and hangs up during the loop.
while (!mqttClient.connect(broker, 8883))
 {
    // failed, retry
    Serial.print(".");
    Serial.println(mqttClient.connectError());
    delay(5000);
  } 

It seems like a problem in the mqttClient.connect(broker,8883). 
After debugging the Sourcecode of NBSSLClient.cpp and NBClient.cpp it shows, that the retry oft he loop doesnt work, because the connectio over the SIM-Network cant connect to mqtt-broker or the communication of the network ist o slow. (presumable)

It's rather strange that it works once and then never again. 

The problem can you see in the following pictures: 
[image]
[image]

measuring buildung: 
- MKR NB 1500
- Antenna
- LiPo 3,7V

sandeepmistry, 2019-06-24:
Hi @splitout,

Can you please enable debug output in the MKRNB lib and copy paste the output of the serial monitor?

To do so you'll need to change the `NB nbAccess;` line in the sketch to `NB nbAccess(true);`.

What SIM card and provider are you using?

splitout, 2019-06-24:
Hi @sandeepmistry,

Iam using the SIM card provider 1NCE. 
the debug mode was enabled. Here the output:

******Connecting to Cellular Network - STARTED******
AT

OK
AT+CMEE=0

OK
AT+CFUN=0

OK
AT+CPIN?

+CPIN: READY

OK
AT+CMGF=1

OK
AT+UDCONF=1,1

OK
AT+CTZU=1

OK
AT+CGDCONT=1,"IP",""

OK
AT+UAUTHREQ=1,0

OK
AT+CFUN=1

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CEREG?

+CEREG: 0,5

OK
AT+CGATT=1

OK
AT+CGACT?

+CGACT: 1,1

OK
******Connecting to Cellular Network - COMPLETED******
 
RadarIOT.azure-devices.net/testdevice/?api-version=2018-06-30
SharedAccessSignature sr=RadarIOT.azure-devices.net%2Fdevices%2Ftestdevice&sig=XXXXXXXXX&se=XXXXXXX
VOR Connect
Try MQTT
Attempting to connect to MQTT broker: RadarIOT.azure-devices.net 
AT+USECMNG=0,0,"AddTrust_External_CA_Root",1082

>
+USECMNG: 0,0,"AddTrust_External_CA_Root","1D355404857803F42424DBF20730A3F"

OK
AT+USECMNG=0,0,"Baltimore_CyberTrust_Root",891

>
+USECMNG: 0,0,"Baltimore_CyberTrust_Root","ACB694A59C170D791529BB19706A6E4"

OK
AT+USECMNG=0,0,"COMODO_RSA_Certification_Authority",1500

>
+USECMNG: 0,0,"COMODO_RSA_Certification_Authority","1B31B071403CC143691ADC43EFDEC18"

OK
AT+USECMNG=0,0,"DST_Root_CA_X3",846

>
+USECMNG: 0,0,"DST_Root_CA_X3","410352DC0FF7501B160028EBA6F45C5"

OK
AT+USECMNG=0,0,"DigiCert_High_Assurance_EV_Root_CA",969

>
+USECMNG: 0,0,"DigiCert_High_Assurance_EV_Root_CA","D474DE57539B2D39C8583C5C065498A"

OK
AT+USECMNG=0,0,"Entrust_Root_Certification_Authority",1173

>
+USECMNG: 0,0,"Entrust_Root_Certification_Authority","D6A5CED5DDD3E00C13D87921F1D3FE4"

OK
AT+USECMNG=0,0,"Equifax_Secure_Certificate_Authority",804

>
+USECMNG: 0,0,"Equifax_Secure_Certificate_Authority","67CB9C013248A829BB2171ED11BECD4"

OK
AT+USECMNG=0,0,"GeoTrust_Global_CA",856

>
+USECMNG: 0,0,"GeoTrust_Global_CA","F775AB29FB514EB775EFF053C998EF5"

OK
AT+USECMNG=0,0,"GeoTrust_Primary_Certification_Authority_G3",1026

>
+USECMNG: 0,0,"GeoTrust_Primary_Certification_Authority_G3","B5E8343C910445848706D2E83D4B805"

OK
AT+USECMNG=0,0,"GlobalSign",958

>
+USECMNG: 0,0,"GlobalSign","9414777E3E5EFD8F0BD41B0CFE7D030"

OK
AT+USECMNG=0,0,"Go_Daddy_Root_Certificate_Authority_G2",969

>
+USECMNG: 0,0,"Go_Daddy_Root_Certificate_Authority_G2","803ABC22C1E6B8D9B3B274A321B9A01"

OK
AT+USECMNG=0,0,"VeriSign_Class_3_Public_Primary_Certification_Authority_G5",1239

>
+USECMNG: 0,0,"VeriSign_Class_3_Public_Primary_Certification_Authority_G5","CB17E431673EE209FE45573F30AFA1C"

OK
AT+USECMNG=2,0,"AmazonRootCA1"

ERROR
AT+USECMNG=0,0,"Starfield_Services_Root_Certificate_Authority_G2",1011

>
+USECMNG: 0,0,"Starfield_Services_Root_Certificate_Authority_G2","173574AF7B611CEBF4F93CE2EE40F92"

OK
AT+USOCR=6

+USOCR: 0

OK
AT+USOSEC=0,1,0

OK
AT+USECPRF=0,0,1

OK
AT+USOCO=0,"RadarIOT.azure-devices.net",8883

OK
AT+USOWR=0,229,"messageASDASD"

ERROR
AT+USOCL=0
.0
3
Retry

OK

Thank you for your help.

splitout, 2019-06-27:
After the connecton to cellular network I tried to get the time with: Serial.println(getTime());
AT+CCLK?
+CCLK: "80/01/06,00:00:20"
Then again after trying connecting to MQTT:

Attempting to connect to MQTT broker: RadarIOT.azure-devices.net 
AT+USECMNG=0,0,"AddTrust_External_CA_Root",1082
>
+USECMNG: 0,0,"AddTrust_External_CA_Root","1D3554048578B03F42424DBF20730A3F"
OK
AT+USECMNG=0,0,"Baltimore_CyberTrust_Root",891
>
+USECMNG: 0,0,"Baltimore_CyberTrust_Root","ACB694A59C17E0D791529BB19706A6E4"
OK
AT+USECMNG=0,0,"COMODO_RSA_Certification_Authority",1500
>
+USECMNG: 0,0,"COMODO_RSA_Certification_Authority","1B31B0714036CC143691ADC43EFDEC18"
OK
AT+USECMNG=0,0,"DST_Root_CA_X3",846
>
+USECMNG: 0,0,"DST_Root_CA_X3","410352DC0FF7501B16F0028EBA6F45C5"
OK
AT+USECMNG=0,0,"DigiCert_High_Assurance_EV_Root_CA",969
>
+USECMNG: 0,0,"DigiCert_High_Assurance_EV_Root_CA","D474DE575C39B2D39C8583C5C065498A"
OK
AT+USECMNG=0,0,"Entrust_Root_Certification_Authority",1173
>
+USECMNG: 0,0,"Entrust_Root_Certification_Authority","D6A5C3ED5DDD3E00C13D87921F1D3FE4"
OK
AT+USECMNG=0,0,"Equifax_Secure_Certificate_Authority",804
>
+USECMNG: 0,0,"Equifax_Secure_Certificate_Authority","67CB9DC013248A829BB2171ED11BECD4"
OK
AT+USECMNG=0,0,"GeoTrust_Global_CA",856
>
+USECMNG: 0,0,"GeoTrust_Global_CA","F775AB29FB514EB7775EFF053C998EF5"
OK
AT+USECMNG=0,0,"GeoTrust_Primary_Certification_Authority_G3",1026
>
+USECMNG: 0,0,"GeoTrust_Primary_Certification_Authority_G3","B5E83436C910445848706D2E83D4B805"
OK
AT+USECMNG=0,0,"GlobalSign",958
>
+USECMNG: 0,0,"GlobalSign","9414777E3E5EFD8F30BD41B0CFE7D030"
OK
AT+USECMNG=0,0,"Go_Daddy_Root_Certificate_Authority_G2",969
>
+USECMNG: 0,0,"Go_Daddy_Root_Certificate_Authority_G2","803ABC22C1E6FB8D9B3B274A321B9A01"
OK
AT+USECMNG=0,0,"VeriSign_Class_3_Public_Primary_Certification_Authority_G5",1239
>
+USECMNG: 0,0,"VeriSign_Class_3_Public_Primary_Certification_Authority_G5","CB17E431673EE209FE455793F30AFA1C"
OK
AT+USECMNG=2,0,"AmazonRootCA1"
ERROR
AT+USECMNG=0,0,"Starfield_Services_Root_Certificate_Authority_G2",1011
>
+USECMNG: 0,0,"Starfield_Services_Root_Certificate_Authority_G2","173574AF7B611CEBF4F93CE2EE40F9A2"
OK
AT+USOCR=6
+USOCR: 0
OK
AT+USOSEC=0,1,0
OK
AT+USECPRF=0,0,1
OK
AT+USOCO=0,"RadarIOT.azure-devices.net","8883",1
OK
AT+USOWR=0,225,"10DE0100044D51545404C2003C00086573707465737431003B5261646172494F542E617A7572652D646576696365732E6E65742F65737074657374312F3F6170692D76657273696F6E3D323031382D30362D3330008B5368617265644163636573735369676E61747572652073723D5261646172494F542E617A7572652D646576696365732E6E65742532466465766963657325324674657374646576696365267369673D675438754235316250594C5541684B3630485A727137795655755930616F67754644784741366F4F7935302533442673653D31353631373030323831"
ERROR
getTime
AT+CCLK?
+CCLK: "80/01/06,00:00:24"

Do you have any idea why it might be?

splitout, 2019-06-28:
Attempting to connect to the cellular network
AT

OK
AT

OK
AT+CMEE=0

OK
AT+CFUN=0

OK
AT+CPIN?

+CPIN: READY

OK
AT+CMGF=1

OK
AT+UDCONF=1,1

OK
AT+CTZU=1

OK
AT+CGDCONT=1,"IP",""

OK
AT+UAUTHREQ=1,0

OK
AT+CFUN=1

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

[... 3253 lines cut ...]
+USORD: 0,""

OK

+UUSORD: 0,536
AT+USORD=0,512

+USORD: 0,512,"303A0603551D1F04333031302FA02DA02B8629687474703A2F2F63726C332E64696769636572742E636F6D2F4F6D6E69726F6F74323032352E63726C303D0603551D200436303430320604551D2000302A302806082B06010505070201161C68747470733A2F2F7777772E64696769636572742E636F6D2F435053300D06092A864886F70D01010B0500038201010011FE72063A210A376EC8F5292202629E450EEDFA069FEFBF9E243517FE3988B2BB01E178DFD18B97FDB777F18A724F493AE8B329D77487E4B9F317608788925582BDD97D13487F06903A8966E79D96F795CD600FD7E1CAF676EEFA9B3548A1987DD2932A4F8116C7A896BBD125F3A66F50DA009B2AA8F43500E3FD2C3BEDE5159B7F160FC78EC9362094F01D01B91250509981E028422C8E78FFEDC69ED5E4CAE9A0AAB3484FA45FE48B18FF13022D9011361866283A229ADE95F159EE9F6D2A17759350C30512D5DE64B4B6BCDBD77659FEA5D6E556E1FE465C3B8AF8EC0EFBEF129091CE201281BA3C52E37FE340F40C4120668718DA3269E0F6DC8F28F2BD0C0001690300186104D360EB863200F624ABEC7E7927ED03EB5CAEFE35F5B6332FD530F82D31AC0CF3C12B59BBCD930C6FCD858B774B7F3DA86F061355776ED76678DCFD75BC0AD0149624A86B63842BE0AC1A02BB356AB7E05F8495D1964A0D65A0D6846B93266BD9020101000BD55AB9"
OK
.-2

+UUSORD: 0,24
AT+USORD=0,512

+USORD: 0,284,"DE4D1890E2F9E8CFBA48037ACA9E6752F94A8F63DAC49E1140DC6A61F67E66852103B36A072C5232A1FA9E86F472AE48EDC8C2F2BCBCBFA7A51162F95BAFFA72228365DC22AF05E9B3242DB61DFE7F7341518F7A7095D2BE414246D784867B06BEF378F3D66E5B17717441D0CEBA5167F8F9958CB27343059752D00F99C0A0A4C11FC7AF1E778B4232BB376DF5D7556420969F1E8896341E5713ED3B0DDC257A3C3C038E0C8B556E6FAFFA44E6D5CC53C88BAEAE16FD01E854ACFF7F72F41F85EBF26A60D4E954DD11E163A5DAABA428959B06639814D3FB7DFA91B0608B6D9A636B39DB478B2452A855A0F0ACB2C15CBCA1272E9898B66BC814567D0D0000180301024000100401050102010403050302030601060300000E000000"
OK
AT+USOCL=0

OK

any reasons?

I used X.508 but on Azure still using X.509?!?!

Thanks

PRAhit, 2019-07-16:
Can you try the below code and let us know the update of the serial monitor with enable debugging by
 NB nbAccess(true);

Please make sure that you are löog in with your credentials on Azure account with Nb 1500 device. 
/*
  Azure IoT Hub NB

  This sketch securely connects to an Azure IoT Hub using MQTT over NB IoT/LTE Cat M1.
  It uses a private key stored in the ATECC508A and a self signed public
  certificate for SSL/TLS authetication.

  It publishes a message every 5 seconds to "devices/{deviceId}/messages/events/" topic
  and subscribes to messages on the "devices/{deviceId}/messages/devicebound/#"
  topic.

  The circuit:
  - MKR NB 1500 board
  - Antenna
  - SIM card with a data plan
  - LiPo battery

  The following tutorial on Arduino Project Hub can be used
  to setup your Azure account and the MKR board:

  https://create.arduino.cc/projecthub/Arduino_Genuino/securely-connecting-an-arduino-nb-1500-to-azure-iot-hub-af6470

  This example code is in the public domain.
*/

#include <ArduinoBearSSL.h>
#include <ArduinoECCX08.h>
#include <utility/ECCX08SelfSignedCert.h>
#include <ArduinoMqttClient.h>
#include <MKRNB.h>
#include <NBUdp.h>

#include "arduino_secrets.h"

/////// Enter your sensitive data in arduino_secrets.h
const char pinnumber[]   = SECRET_PINNUMBER;
const char broker[]      = SECRET_BROKER;
String     deviceId      = SECRET_DEVICE_ID;

NB nbAccess;
GPRS gprs;

NBUDP Udp;

NBClient      nbClient;            // Used for the TCP socket connection
BearSSLClient sslClient(nbClient); // Used for SSL/TLS connection, integrates with ECC508
MqttClient    mqttClient(sslClient);

const char* timeServer = "time.nist.gov";
const int NTP_PACKET_SIZE = 48;

unsigned long lastMillis = 0;

void setup() {
  Serial.begin(9600);
  while (!Serial);

  if (!ECCX08.begin()) {
    Serial.println("No ECCX08 present!");
    while (1);
  }

  // reconstruct the self signed cert
  ECCX08SelfSignedCert.beginReconstruction(0, 8);
  ECCX08SelfSignedCert.setCommonName(ECCX08.serialNumber());
  ECCX08SelfSignedCert.endReconstruction();
[... 124 lines cut ...]
  Serial.println("Publishing message");

  // send message, the Print interface can be used to set the message contents
  mqttClient.beginMessage("devices/" + deviceId + "/messages/events/");
  mqttClient.print("hello ");
  mqttClient.print(millis());
  mqttClient.endMessage();
}

void onMessageReceived(int messageSize) {
  // we received a message, print out the topic and contents
  Serial.print("Received a message with topic '");
  Serial.print(mqttClient.messageTopic());
  Serial.print("', length ");
  Serial.print(messageSize);
  Serial.println(" bytes:");

  // use the Stream interface to print the contents
  while (mqttClient.available()) {
    Serial.print((char)mqttClient.read());
  }
  Serial.println();

  Serial.println();
}

splitout, 2019-07-22:
Hi @PRAhit, 

actually I tested your example. 
Now I got the Error-Code  (-2) shows txt-file and example.ino

The ECCX08 is active. 
Please enter the issue year of the certificate? (2000 - 2031) [2019]: 2019
Please enter the issue month of the certificate? (1 - 12) [1]: 6
Please enter the issue day of the certificate? (1 - 31) [1]: 14
Please enter the issue hour of the certificate? (0 - 23) [0]: 12
Please enter how many years the certificate is valid for? (1 - 31) [31]: 31
What slot would you like to use for the private key? (0 - 4) [0]: 0
What slot would you like to use for storage? (8 - 15) [8]: 8
Would you like to generate a new private key? (Y/n) [Y]: Y

The SHA1 key is deposited to azure device on primary and secondary thumbprint. 

After the first try to connect with the mqtt broker, the loop oft he mqtt connect freezes again.  

The AT-commands: 
Attempting to connect to the cellular network
AT

OK
AT+CMEE=0

OK
AT+CFUN=0

OK
AT+CPIN?

+CPIN: READY

OK
AT+CMGF=1

OK
AT+UDCONF=1,1

OK
AT+CTZU=1

OK
AT+CGDCONT=1,"IP",""

OK
AT+UAUTHREQ=1,0

OK
AT+CFUN=1

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

+CEREG: 0,0

OK
AT+CEREG?

[... 4480 lines cut ...]

+USORD: 0,""

OK
AT+USORD=0,512

+USORD: 0,""

OK

+UUSORD: 0,536
AT+USORD=0,512

+USORD: 0,512,"303A0603551D1F04333031302FA02DA02B8629687474703A2F2F63726C332E64696769636572742E636F6D2F4F6D6E69726F6F74323032352E63726C303D0603551D200436303430320604551D2000302A302806082B06010505070201161C68747470733A2F2F7777772E64696769636572742E636F6D2F435053300D06092A864886F70D01010B0500038201010011FE72063A210A376EC8F5292202629E450EEDFA069FEFBF9E243517FE3988B2BB01E178DFD18B97FDB777F18A724F493AE8B329D77487E4B9F317608788925582BDD97D13487F06903A8966E79D96F795CD600FD7E1CAF676EEFA9B3548A1987DD2932A4F8116C7A896BBD125F3A66F50DA009B2AA8F43500E3FD2C3BEDE5159B7F160FC78EC9362094F01D01B91250509981E028422C8E78FFEDC69ED5E4CAE9A0AAB3484FA45FE48B18FF13022D9011361866283A229ADE95F159EE9F6D2A17759350C30512D5DE64B4B6BCDBD77659FEA5D6E556E1FE465C3B8AF8EC0EFBEF129091CE201281BA3C52E37FE340F40C4120668718DA3269E0F6DC8F28F2BD0C0001690300186104EA31D7EBEEC5FBAE83F8428E1BF2A9D0DB02FC3E006BF31B35E921BDA899E408D3A6256CA5799BDDC78D3D47C1C40BB685B50FB64FC4962DCC1B6500B7DF7BFA22C8E0B6D8873A0DA33C55C349371B4F0B1D2D15E7B6577C1E9D208FF3B39643020101004A68A292"
OK

+UUSORD: 0,24
.-2
AT+USORD=0,512

+USORD: 0,284,"D3874023E1E40F12B884FFDF0380D963689F66C068436E26ECAEA427EA7B46FC230C13B5962ABD8FC534A340F0EF9B0584C599B07E83FAE2A240E85A1C3C0FB2F7DFF1D675F5B2D1ED2E232518C0B546290654AB5DE22DFB9DE5CF342FE4EAA84CE41DAF3311CA04DC70E8E96D0AE52232D5F7E4F517C3104D21D2D37E1FD40084201450627DC1EEBFFB3D5805F87CD44990F4199ABF3782F70907121E4A4FCF1A176E4C8ED376533CB8351B7ADFC22A9ED571FDF494E29BC8CCCED15E1E95B97FB8335D72DACE60BD63B0039CE23F697E79FCAC59852A02B91006949D7E8AC943F59436FBAB9A36658180443162B521EC4CC34FB9220A096367F8910D0000180301024000100401050102010403050302030601060300000E000000"
OK
AT+USOCL=0

OK

DanielMDias, 2019-11-12:
Seems there is a problem with some carriers.
If you set NB nbAccess=true;
and on function getTime() you   Serial.println(nbAccess.getTime());
You will find there is printed the following info:

AT+CCLK?
+CCLK: "80/01/06,00:00:22"

Since the timestamp returned from carrier is older than the used when generating the certicate will make the error -2 = connection refused.

https://www.unixtimestamp.com/
One workaround is making the function getTime() return the timestamp from the above page (Not Recommended! for production).

Hope this helps
Daniel

sandeepmistry, 2019-11-14:
You could also try to change the sketch to retrieve the time using NTP over UDP: https://github.com/arduino-libraries/MKRNB/blob/master/examples/GPRSUdpNtpClient/GPRSUdpNtpClient.ino

splitout, 2019-11-15:
The problem should have been solved months ago. Not month later.
The whole solution of your support did not work!
Fix the bug...

cabsteena, 2020-02-03:
I had the same issue and it would work perfectly then not??
I think it was disconecting from the network every now and then due to poor signal, and if it happened whilst trying to connect to MQTT it would crash in the while (!mqttClient.connect(broker, 8883)) part of the void mqtt() code.
I added this to the bottom of the mqtt() while code:

 if (nbAccess.status() != NB_READY || gprs.status() != GPRS_READY) {
    connectNB();
  } 

Try changing your void connectMQTT();
to this:

while (!mqttClient.connect(broker, 8883)) {
    // failed, retry
    Serial.print(".");
    Serial.println(mqttClient.connectError());
    delay(5000);
    if (nbAccess.status() != NB_READY || gprs.status() != GPRS_READY) {
    connectNB();
  } 
  }

Hope that helps.