arduino-libraries/MKRNB #23: MKR NB1500 doesn't work with AWS MQTT

ravelab, 2019-07-31:
Hi,

I'm using ArduinoBearSSL. I am able to get MKR1000USB (wifi) to work with AWS MQTT server, and MKR NB1500 to work with Google MQTT server . However, I am unable to get MR NB1500 to work with AWS MQTT server using the "AWS IoT NB" example. It is stuck at 'AT+USOCO=0,"XXX.iot.us-east-2.amazonaws.com",8883' and timeout after 5 minutes:

Attempting to connect to MQTT broker: XXX.iot.us-east-2.amazonaws.com
AT+USOCR=6
+USOCR: 0

OK
AT+USOCO=0,"XXX.iot.us-east-2.amazonaws.com",8883
ERROR

+UUSOCL: 0
AT+USOCL=0
ERROR
0000-00-00T00:00:00Z MQTT connect error: -2

Any ideas? Thanks

sandeepmistry, 2019-08-02:
Hi @ravelab,

`XXX.iot.us-east-2.amazonaws.com` does not appear to be a valid host name to me:

ravelab, 2019-08-24:
Hi @sandeepmistry yeah XXX is not an actual address. I followed those instructions, I can connect to aws via wifi, but not nb. I could connect to azure or gcp via nb though. We think it might be a carrier issue. Thanks

closed 2019-08-24 by ravelab

ravelab, 2020-03-07:
@zvikarp for me it was a ublox modem firmware issue

Ar00001, 2020-07-10:
Hi @ravelab I am also facing the exact same problem, could You please describe maybe a little bit how You fix the modem firmware issue? Did You update the Firmware of the U Blox Module? And how did You update? Thanks for Your help.

marcelchelo, 2022-10-17:
@Ar00001   Below is a link that show how to update the modem.   You have to contact u-blox  to request the later firmware. 

https://forum.arduino.cc/t/firmware-upgrade-for-ublox-sara-r410m-02b-on-the-mkr-nb-1500-2/699292/61?_gl=1*fexqls*_ga*MTI2MDk2ODE3NS4xNjY1NDM3NTIw*_ga_NEXN8H46L5*MTY2NTk3ODI5NC41LjEuMTY2NTk4MTA3NS4wLjAuMA..

mrks89, 2023-01-11:
> To connect to AWS I would suggestion the following:
> 
> Follow the first part of https://create.arduino.cc/projecthub/132016/securely-connecting-an-arduino-mkr-wifi-1010-to-aws-iot-core-a9f365 to setup the private key and public certificate
> Run the following sketch, instead of the WiFi one mentioned in the link above: https://github.com/arduino/ArduinoCloudProviderExamples/tree/master/examples/AWS%20IoT/AWS_IoT_NB

the links do not work anymore. Any idea how to do so?

per1234, 2023-01-16:
@ravelab I have updated the link. The tutorial is now here:

https://docs.arduino.cc/tutorials/mkr-wifi-1010/securely-connecting-an-arduino-mkr-wifi-1010-to-aws-iot-core

If you have any other questions, please ask on the Arduino Forum:

https://forum.arduino.cc/

I'm sure we'll be able to help you out over there.