arduino-libraries/MKRNB #27: Update firmware on mkr1500 UBLOX SARA-R410M-02B

rubenfrolic, 2019-09-27:
Hi,

I recently bought a Arduino mkr1500 but it came with obsolete firmware on the UBLOX SARA-R410M-02B. 

firmware version: L0.0.00.00.05.06 [Feb 03 2018 13:00:41]

I'd like to update the firmware to the latest and was wondering how to do this? Is there some more info on how to achieve this?

Thx

rubenfrolic, 2019-12-03:
Hi,

I've been testing for a while now with the arduino MKR1500 and ran into some issues concerning the registration on the LTE-m and NB-IOT network. I basically tried to figure out what network the Ublox r410m module was connected to.

With the AT+URAT command you can set your preferences for 1st and 2nd network: LTE-m over NB-IOT, visa versa, or just one of the two. Then with the AT+COPS command you can see what network the module is actually connected to.

Whenever I did this, I however got unexpected answers. Based on the UBlox datasheet, I should either get a 7 (LTE-m) or 9 (NB-IOT) based on the network. I however got returned an 8, something that according to Ublox would be impossible.

After a lot of time debugging and testing I figured out it has to do with the Ublox firmware. The module comes with version 05.06 while the current latest version provided by Ublox is 05.08.

Between these versions Ublox has changed some references and commands. So when you work with the mkr1500 it's good to know you can't 100% rely on the lastes version of the AT-command manual that Ublox provides. And it's also good to know that arduino ships the modules not with the latest firmware and that if this changes in the future, this might have implications for your firmware on your arduino.

Also would it be possible from Arduino's perspective to provide an AT manual which corresponds with the firmware version 05.06 which arduino ships the modules with? Also since it's not possible to update the firmware on the Ublox module (like for instance is with the SODAQ r410m)

B-Clever, 2020-05-22:
I updated the firmware of my NB 1500 following these instructions a few months ago:

https://forum.arduino.cc/index.php?topic=628808.0

But take your time and read the instructions carefully! 
You have to upload the Passthrough sketch to the Arduino first. 
Follow comment 6, 7 and 13 of user "zbelding" on the linked page.

CptHolzschnauz, 2020-08-20:
This is a guide with windows for the modems with no bad block:

1. Load the Sara pass-through sketch. Sketch Examples->MKRNB->Tools>SerialSARAPasstrough

Remember the Port Number

2. Close arduino IDE

3. Open Extra PuttyTel under Windows (It's the easiest way)

4. Set com port, baud, and "Serial" radio button

5. On the left click on "Terminal" and select both "Force on" radio buttons

6. Click "Open" to start terminal

7. Send AT+UFWUPD=3

8. After modem replies: "+UFWUPD: ONGOING", choose X-modem-1k transfer from file transfer menu. This can take some seconds.

9. Choose 0bb_stg1_pkg1_0m_L56A020X_to_L58A0204.bin and begin transfer

10. After complete, send AT+UFWINSTALL

10.5. After OK, wait 10 sec. minimum

11. Send AT+UFWSTATUS?

12. Compare results to Readme

13. Depending on results you may have to try another stg1 package

55436F6D flag means Update has successfully completed

If it failed andyou see +UFWSTATUS: 55457272, 19a, ffe3, then you need to go back to Step 1 and try another stg1 boot package.

In case of success:

14. Send AT+UFWUPD=3

15. After modem replies: "+UFWUPD: ONGOING", choose X-modem-1k transfer from file transfer menu

16. Choose 0bb_stg2_L56A020X_to_L58A0204.bin and begin transfer

17. After complete, send AT+UFWINSTALL

18. wait 21 mins

19. Send ATI9 to make sure it worked

 

ati9

 L0.0.00.00.05.08,A.02.04

realonedoe, 2020-08-30:
Use m-center from u-blox and make the firmware update about FOTA (Firmware over the air). 

**This is a very useful software to test, configure and update your u-blox-module!**

https://www.u-blox.com/en/product/m-center#tab-documentation-resources

1. Load the Sara pass-through sketch. Sketch Examples->MKRNB->Tools>SerialSARAPasstrough
Remember the Port Number

2. Close arduino IDE

3. Start m-center
3.1 Choose the right port
3.2 Go to FOTA (Firmware over the air)

Please read the manual before.

CptHolzschnauz, 2020-08-31:
I did, with no sucess. 
An internal PDP context should be activated in order to run FOTA application
is what i get. 
But it's a great hint that the the m-center can be connected trough passtrough. Even the techie from u-blox wrote me that it's only possible trough a soldered USB cable...So easy. For me, first light with your post!
Thx a bunch

CptHolzschnauz, 2021-04-04:
There is a new firmware with a new and wicked update procedure:
https://forum.arduino.cc/index.php?topic=731275.0
Try to have fun!