arduino-libraries/MKRNB #28: NB::getLocalTime() and NB::getTime() Incorrectly Assumes Non GMT Time From CCLK

bleckers, 2019-10-23:
We're testing with the MKRNB 1500 and the Telstra network in Australia (in ADST/AEST timezone; currently +11) and the CCLK command in the get time functions of the MKRNB libraries actually return a GMT time rather than an assumed local time by the looks of things.

In the NB::getTime() function, it converts the CCLK value with an offset of -11, which when the CCLK command returns a GMT value in our case, it is actually offset by -22 hours from current time (GMT time minus -11).

For example we got this response at 2111 on the 23/10 (+44 in this case is +11hrs which gets subtracted from the time in the code to produce "GMT", when in actual fact it's already GMT):
`+CCLK: "19/10/23,10:11:41+44"`

In the NB::getLocalTime() function, it doesn't change the value based on the TZ value, but in our case it returns GMT.

Looking at the SARA R4 AT commands, it doesn't specify what time format the CCLK value actually returns in, so we can never actually assume it's the current local time (with GMT offset applied). Also it states that TZ is network dependent and might not actually return a TZ.

With all this in mind, it seems like there's 4 different network time formats depending on the provider, neither of which can be determined with the information from the radio alone:

1. CCLK returns GMT +/- TZ
2. CCLK returns GMT without TZ
3. CCLK returns local time +/- TZ
4. CCLK returns local time without TZ

So the NB::getLocalTime() shouldn't really be used as it "might" not actually apply the correct offset. getTime should be used instead, but to solve the above ambiguity, you need to provide your own smarts based on which network provider you are using and which values are returning (i.e. a lookup table based on provider and location to determine what your time should actually be).

Unless there's an AT command I'm missing that might solve this problem?

janakelarsson, 2020-01-17:
Also here in Sweden, on the other side of the globe, on Halebop SE, does CCLK return GMT +/- TZ.

janakelarsson, 2020-01-17:
I'm using the code from the shifttime branch in my personal fork of the library

janakelarsson, 2020-01-20:
> Also here in Sweden, on the other side of the globe, on Halebop SE, does CCLK return GMT +/- TZ.

I mean, CCLK returns GMT, and then tells which timezone shift to apply to get local time.

janakelarsson, 2020-01-20:
Init code has `MODEM.send("AT+CTZU=1");` which corresponds to `1: automatic time zone update via NITZ enabled; if the network supports the service, update the local time to the module (not only time zone)` according to the manual. My operator obviously gives the right TZ, so maybe this mode is just for adjusting the TZ of the response, and not the time value returned itself?

gasagna, 2022-08-15:
I have a MKR NB 1500, with a SARA-R410-02B device. According to the device's [docs](https://content.u-blox.com/sites/default/files/SARA-R4_ATCommands_UBX-17003787.pdf) (bottom of page 42), the `AT+CCLK?` command always returns UTC time, "regardless of the current +CTZU AT command settings".