arduino-libraries/MKRNB #30: How to write more that 512 bytes to a socket

mnelson920, 2019-11-14:
From what I seeing this library writes data to a socket using HEX. Doing this limits the modem to write 512 bytes. I'm trying to write a JSON message that is 570 bytes to AWS. Is there any way to change this limitation?

sandeepmistry, 2019-11-14:
Hi @mnelson920,

Please share a sketch to reproduce any issues you are facing.

`NBClient` has a loop now to chunk sending data over a socket that is over 512 bytes: https://github.com/arduino-libraries/MKRNB/blob/master/src/NBClient.cpp#L266-L329

CptHolzschnauz, 2021-07-22:
I had timing problems with the loop so i suggest (because the modem can do it):
https://github.com/arduino-libraries/MKRNB/pull/75

CptHolzschnauz, 2021-08-22:
Hi @sandeepmistry  : But it's 256 bytes per push, it's more stable with 512

https://github.com/arduino-libraries/MKRNB/pull/84

Also the asynchronous socket close is a game changer in stability