arduino-libraries/MKRNB #31: MKR1500 AT Serial Commands only work from time to time...frustration

njordan77, 2019-12-15:
Hi,
i've an MKR1500, i installed the passthrough sketch which from time to time works so that AT results in an OK. A lot is written in the internet about unstable behavior but i could not find any solution and its highly frustrating as from 50 tries maybe it works once that AT commands result in reaction. Looks like the modem is not reached at all.

As a lot of other people have the same issues i'm surprised that this is not any open issue here....i hoped that there needs to be done a slightly change (delay,...) in the sketch to make it work stable.....

I did already add a LiPo Battery,...nothing improved the bad experience i have right now. Thanks a lot Norbert

Rocketct, 2020-01-23:
Hi @njordan77 i looking on it, have you any particular AT commands sequence to make the issue happens? how much command you have to send to the nb in order to hang it? can you share the results of the command ATI9?

thekunalsaini, 2020-03-20:
@Rocketct I fix some code in nb.cpp please refer that

janakelarsson, 2020-03-25:
(@thekunalsaini didn't fix any code in NB.cpp.) 

He refers to my pull request #44 that removes an extraneous poll(), I doubt that helps.

I had similar trouble because of a bad connection of the USB connector, the serial connection between computer and NB 1500 was glitchy. I'm sure you checked this already, but doesn't hurt to mention it.

thekunalsaini, 2020-03-26:
@janakelarsson yeah:)

Rocketct, 2020-03-26:
thanks guys, @janakelarsson yes thank you for the help :) i think that could be useful understand in which condition @njordan77 is, i mean which command went used, core, library and IDE version, if he try to change ports and operative system, which battery is using, if this problem born spontaneously or after the use of some examples, if he was able to use the other sketches behind, practically we need to get the same condition to investigate over it