arduino-libraries/MKRNB #35: Extra MODEM.poll() in NB.cpp

janakelarsson, 2020-01-23:
Hi, in NB.cpp on line 153 I find `int ready = MODEM.ready();`, this calls `MODEM.poll()` internally to update the internal state, then reads and returns the state. What is the reason for the extra `MODEM.poll();`on line 159?

I have commented it out in my copy of the library. The code runs fine without it, and it is not present in the parallel library for MKR 1400 GSM.

tp-Fernando, 2020-02-24:
Hey my friend. I've been using this library and noticing some strange behaviour. Very unstable.
Did comment out the extra modem.poll improve things? Thanks

janakelarsson, 2020-02-24:
Since I do not know what strange behavior you saw I cannot really say. In any case, I did not run the library long enough before deleting that line to be able to say anything specific. 

My application runs fine, but it took a number of tries to get the calls right.

A recommendation is that you make sure to check that the modem is ready before attempting to perform any action. Doing something new before the modem is finished with the previous action will trigger an error, and that will look to you as if the thing is very unstable.

aentinger, 2020-03-23:
Hi @janakelarsson :wave: Thank you for uncovering this. I agree that the additional `MODEM.poll()` on line 155 is not necessary. It seems that @thekunalsaini has already created pull request #43 to deal with this issue.

Normally I'd just go ahead and merge it in but I'm really not happy with the way this particular user is trying to score "cheap" commits in the Arduino repositories and claiming to have performed a real test with hardware when this seems quite unlikely. Normally I'd give the benefit of doubt but in that particular case such coincidences have been accumulating.

Now, we at Arduino want to award those who take the time and effort for performing meaningful contributions so if you were to create a PR with a meaningful title and commit message I'll be more than happy to merge it in.

janakelarsson, 2020-03-23:
WIll do, tonight

thekunalsaini, 2020-03-23:
@aentinger sir firstly I have not cheap commits basically I have mention this user and after proper chdcking I made the commits since in issue @janakelarsson not made commit and this issue is open for all.
That is why I made commit 
As I love Arduino community so do not say like this sir and I mention this issue if I made this request without mentioning then,
I prefer that its my responsihility to mention

Thank you,

janakelarsson, 2020-03-23:
You didn't mention the user (me) and didn't reference the issue, I also doubt you tested on actual hardware.

PR #44 (2020-03-24): Remove extraneous poll()
There is an extra poll() in NB.cpp, this pull request removes that, closes #35 .
closed 2020-03-24 by aentinger