arduino-libraries/MKRNB #45: SMS are received only after boot or after sending SMS

janakelarsson, 2020-03-26:
I use my NB 1500 to monitor a sensor, and send SMS if a warning threshold is reached. The code also monitors incoming SMS for sensor level queries, and responds when this happens. Used to work well. Recently, the code has stopped receiving SMS, unless a) the NB1500 is rebooted, or b) a warning SMS is sent, and only then it receives pending SMSes. 

The code issues AT+CMGL="REC UNREAD" once per second and receives "OK" (I can see this with debug mode on). When an SMS is sent, the next query returns pending received SMSes.

Is anyone else seeing this? Is there some configuration I've changed without meaning to? Has this been caused by a hard reboot at the wrong moment? How do I return to the old behavior?

janakelarsson, 2020-04-11:
Have now added code to restart the connection with a longer repetition. This works a workaround, for now.

janakelarsson, 2020-06-02:
The frequent restarts may have bricked the device. See issue #47

janakelarsson, 2020-07-23:
Obtained new device, fw update, and rewritten powerup and powerdown, closed by #55

closed 2020-07-23 by janakelarsson

CptHolzschnauz, 2020-08-14:
I have exactly the same issue. Modem Firmware updatet, latest MKRNB with your changes compiled, but still got this behavior. You closed the issue because you solved it? If so, how?
Thx

janakelarsson, 2020-08-15:
I can only guess. My best guess is that the behavior is caused by a sudden shutdown, so that receiving SMS does not work as expected. My workaround, restarting often, then bricked the device.I then got a new device, immediately updated the firmware, and rewrote the library. So basically the fix was: get a new device, update firmware, use rewritten library.

Note that I needed to change profile of the new device to get it to work properly. I'm in Europe so I used the passthrough sketch to issue the following commands:
- Select network operator "standard Europe":                                                 AT+UMNOPROF=100
- Reboot:                                                                                                          AT+CFUN=15
- Issue AT until you get OK so the modem is rebooted 
- Then _quickly_ turn off powersave, otherwise modem turns off in seconds: AT+CPSMS=0
- Select combined attach, both speech/SMS and data:                                   AT+USVCDOMAIN=2
You can read about this in https://www.u-blox.com/sites/default/files/SARA-R4-Application-Development_AppNote_%28UBX-18019856%29.pdf

CptHolzschnauz, 2020-08-16:
I understand, it's merely impossible to solve e flaky problem from remote...
I followed every step with one result. It gets worse.. Now the operaters name is no more visible
Current carrier: AT+COPS?
 +COPS: 0
The receiving of SMS doesn't work anymore at all (Only after a hard reset with the boards button, then the sketch finds the sms and everything works..)

Do you know a command queue to something like
Factory defaults, set the right settings for receiving only SMS and everything else neccesary?

2nd question: You described your sketch and it sounds similar to mine. Are you running now this sketch relieable on the mkr1500 for a long time?
Thx for your help and the link

janakelarsson, 2020-08-16:
Are you in Europe? Perhaps a different network operator profile would be better.

janakelarsson, 2020-08-16:
Read chapter 7. And https://www.u-blox.com/en/docs/UBX-17003787#%5B%7B%22num%22%3A181%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C59.527%2C239.677%2Cnull%5D. Network operators are in a table on page 65. 

Deutsche Telekom is operator 31, for example.

CptHolzschnauz, 2020-08-16:
Yes,i am. Working with a MobileThings SIM in Switzerland.

I just using the official ReadSMS Sketch vrom the examples section and with a SMS in the dark hole memory and sending a sms immediatly after processing the first one. Works one (or two) times perfektly and then endless
09:07:41.790 -> OK
09:07:42.874 -> AT+CMGL="REC UNREAD"

until rebooting of the board, exactly like you descriped it...
Thank you for helping me, it's so annoying fixing others bugs

09:07:41.790 -> OK
09:07:42.874 -> AT+CMGL="REC UNREAD"

09:07:42.874 -> OK
09:07:43.970 -> AT+CMGL="REC UNREAD"

09:07:43.970 -> +CMGL: 0,"REC UNREAD","+41786025000",,"20/08/16,09:06:41+08"
09:07:43.970 -> S
09:07:43.970 -> 
09:07:43.970 -> OK
09:07:43.970 -> Message received from:
09:07:43.970 -> +41....
09:07:43.970 -> 
09:07:43.970 -> END OF MESSAGE
09:07:44.007 -> AT+CMGD=0

09:07:44.007 -> OK
09:07:44.007 -> MESSAGE DELETED
09:07:45.100 -> AT+CMGL="REC UNREAD"

09:07:45.138 -> OK
09:07:46.227 -> AT+CMGL="REC UNREAD"

09:07:46.227 -> OK
09:07:47.335 -> AT+CMGL="REC UNREAD"

09:07:47.335 -> OK
09:07:48.427 -> AT+CMGL="REC UNREAD"

09:07:48.427 -> OK
09:07:49.518 -> AT+CMGL="REC UNREAD"

09:07:53.952 -> OK
09:07:55.026 -> AT+CMGL="REC UNREAD"

09:07:55.026 -> OK
09:07:56.137 -> AT+CMGL="REC UNREAD"

09:07:56.137 -> OK
09:07:57.237 -> AT+CMGL="REC UNREAD"

09:07:57.237 -> OK
09:07:58.337 -> AT+CMGL="REC UNREAD"

09:07:58.337 -> OK
09:07:59.456 -> AT+CMGL="REC UNREAD"

09:07:59.456 -> +CMGL: 0,"REC UNREAD","+41....",,"20/08/16,09:07:57+08"
09:07:59.456 -> S
09:07:59.456 -> 
09:07:59.456 -> OK
09:07:59.456 -> Message received from:
09:07:59.456 -> +41...
09:07:59.456 -> 
09:07:59.456 -> END OF MESSAGE
09:07:59.495 -> AT+CMGD=0

09:07:59.495 -> OK
09:07:59.495 -> MESSAGE DELETED
09:08:00.577 -> AT+CMGL="REC UNREAD"

09:08:00.577 -> OK
09:08:01.688 -> AT+CMGL="REC UNREAD"

09:08:01.688 -> OK

CptHolzschnauz, 2020-08-16:
Yes i did but what about:
If the <MNO> parameter is set to 1 (SIM ICCID select), the mobile network operator profile is selected according
to the recognized SIM Issuer Identifier Number (IIN). For more details on the recognized SIM IINs ranges, see
the SARA-R4 application development guide 

The SMS are in the modem! I don't think it's about the carrier the sms receiving works perfect with no dely of hours.

I think there must be a bug in the library, because the SMS are stored in the modems memory and marked as read without  passing it over to the sketch. Stuff like that is so annoying..

janakelarsson, 2020-08-16:
Then you should report a bug. I do not have this behavior. (If you send a literal "OK" the library will misbehave. My fix to that issue was not adapted, the patch they used might not work.) You are writing in a closed issue, report this as a new issue.

The procedure is to set profile 1 and then reboot twice:
T+UMNOPROF=1
OK
AT+CFUN=15
OK
AT+UMNOPROF?
+UMNOPROF: 0
OK
AT+CFUN=15
OK
AT+UMNOPROF?
+UMNOPROF: 3
OK

The last answer will of course depend on what it finds. I could not use this, probably because my operator was not in the internal database. (This can be found in the Development App note)

janakelarsson, 2020-08-16:
You can enable debug mode in the library to see what happens. You can even edit the source code of the library to show what happens, which part of the code is used, and so on. This is how I learnt how to modify the library.

CptHolzschnauz, 2020-08-16:
You mean NB nbAccess(true); or is there a deeper debug mode? How to turn on?
As you know, it's hard to debug a code that is working some times and some times not...