arduino-libraries/MKRNB #46: MKR NB 1500 Not working with AWS

ircole, 2020-04-19:
This similar to another post here but a little different.  I have no problem connecting to AZURE but never completes connecting with AWS MQTT.   I've followed the suggestions that I've found but with no success.  I created the csr on the 1500 and copied to AWS.  Maybe this is a firmware issue as was mentioned.  Any help will be appreciated.  Very likely something that I'm missing as I'm not as young as I used to be.

Attempting to MQTT broker: xxxxdo87u8bnm5-ats.iot.us-west-2.amazonaws.com 
AT+USOCR=6
+USOCR: 0
OK

AT+USOCO=0,"xxxxdo87u8bnm5-ats.iot.us-west-2.amazonaws.com",8883
OK

AT+CCLK?
+CCLK: "20/04/19,19:43:47-20"
OK

AT+USORD=0,512
+USORD: 0,""
OK

AT+USOWR=0,251,"16030100F6010000F2030300000000E6D265D85F7FF035C61BDDD9EE39284DD8F68CE386F9AE3211B104A300005ACCA9CCA8C02BC02FC02CC030C0ACC0ADC0AEC0AFC023C027C024C028C009C013C00AC014C02DC031C02EC032C025C029C026C02AC004C00EC005C00F009C009DC09CC09DC0A0C0A1003C003D002F0035C008C012C003C00D000A0100006FFF0100010000000033003100002E61326F7A646F38377538626E6D352D6174732E696F742E75732D776573742D322E616D617A6F6E6177732E636F6D0001000101000D001600140403030305030603020304010301050106010201000A000A0008001700180019001D000B00020100"

+USOWR: 0,251

OK

AT+USORD=0,512

+USORD: 0,""

OK

+USORD: 0,512,"16030314E20200005203035BE8B82F248F528AC6EB3B0D5DAF9D8C34DAF5D7B337F028993E12FBD89F134A209A8D0331E81B1C8EE98928294DF779D109CC98F7B6FD62A3BF78EDB524E9F4E9C02F00000AFF0100010000010001010B00131100130E0005A6308205A23082048AA0030201020210011AD3074D6DB362ACF2690BAD8B1D54300D06092A864886F70D01010B05003046310B3009060355040613025553310F300D060355040A1306416D617A6F6E31153013060355040B130C536572766572204341203142310F300D06035504031306416D617A6F6E301E170D3230303330323030303030305A170D3231303231303132303030305A30283126302406035504030C1D2A2E696F742E75732D776573742D322E616D617A6F6E6177732E636F6D30820122300D06092A864886F70D01010105000382010F003082010A0282010100C800A392D2A7C462D1C664B22EBA4E962A01DB9251CACFF3FFBA4E4C20F8BD78B9E25D0DBCAD9E8B957FA8666610C7D4580F28E40509E4695B6EC54D5EB40E00BE4C99D97E0C4CE923DF480490D1F0B12DB04A91094131A0D7A86A0CA909506113BA8A43C85B034832DF6651BCCCDA5CFCC51D267C747F8B143319A75437E9C2C7D713DADC0650F48FCB7847E3BF8DA31999B4B2F5C244EB8FD3F14EA73AF44AF71DCB6C09DFF5DB6508E854803D80D89F3ECAEE802BB6F88531"

OK

+UUSORD: 0,868

AT+USORD=0,512

+USORD: 0,512,"8BADB656104EFC69C6C3D928EFD7156503C6973F0B5E320F0225E28802CC62C72C6CDC33243FF1A9C0DF4900E9B9CCD2C560164ECF89DDCB1ECB3A110F3C17895498D5F562650203010001A38202A8308202A4301F0603551D2304183016801459A4660652A07B95923CA394072796745BF93DD0301D0603551D0E041604148B20B81211B0162903BD76FBFB58ECC757B1592030450603551D11043E303C821B696F742E75732D776573742D322E616D617A6F6E6177732E636F6D821D2A2E696F742E75732D776573742D322E616D617A6F6E6177732E636F6D300E0603551D0F0101FF0404030205A0301D0603551D250416301406082B0601050507030106082B06010505070302303B0603551D1F043430323030A02EA02C862A687474703A2F2F63726C2E73636131622E616D617A6F6E74727573742E636F6D2F73636131622E63726C30200603551D2004193017300B06096086480186FD6C01023008060667810C010201307506082B0601050507010104693067302D06082B060105050730018621687474703A2F2F6F6373702E73636131622E616D617A6F6E74727573742E636F6D303606082B06010505073002862A687474703A2F2F6372742E73636131622E616D617A6F6E74727573742E636F6D2F73636131622E637274300C0603551D130101FF0402300030820106060A2B06010401D6790204020481F7"

OK

+UUSORD: 0,356

AT+USORD=0,512

+USORD: 0,356,"0481F400F2007700F65C942FD1773022145418083094568EE34D131933BFDF0C2F200BCC4EF164E3000001709CDDAB860000040300483046022100B308999E337B3C345D84B4A20B8372029D4DC2BBEE86B0FE45413AD80F51D0BD022100D737FF8AD4EF825876F271B8B9AF44F9EE19E7808A91071A123D22169141AE4C0077005CDC4392FEE6AB4544B15E9AD456E61037FBD5FA47DCA17394B25EE6F6C70ECA000001709CDDABD10000040300483046022100A47CD089175D65096F3FCEDFADDF33546ADA46366B6F3911043AD0422E510D1402210099271A39E0D783C4B8D758845A56CE64337D4BF481421231E77A734806D9FC20300D06092A864886F70D01010B0500038201010026576F12527FE2AD8AF5685E60E33B620362D38D617BD14EF9680B15570495DF77BA8593FAA26019AC70E8BE4A498B6B6B2A8E7C6AE8596860B41FDF354C8D384025635547C1F1F6E433069DCFB74925BA20D233783CB12F1D"

OK

AT+USORD=0,512
+USORD: 0,""
OK

AT+USORD=0,512
+USORD: 0,""
OK
These reads go on forever.

ircole, 2020-06-23:
The problem was me.  I had copied the wrong security file.

closed 2020-06-23 by ircole

marcelchelo, 2023-01-28:
I'm struggling to connect to aws mqtt, do you have any documentation links that you followed?