arduino-libraries/MKRNB #50: Arduino mkrNB 1500 send sms not working

web-10, 2020-06-22:
My new mkr nb 1500 cannot send sms is not working. But if the number to send the sms to has previously sent a message to the mkr 1500 than it works. It seems that the mkr 1500  remembers the path or so to send an sms if it previously has received one from the sender.

I use the example send sms provided by the Arduino editor version 1.8.12. Examples / examples from libraries / mkrNB / tools / send sms.

Receiving SMS works fine using the Examples / examples from libraries / mkrNB / tools / receiving sms: works fine of Arduino IDE 1.8.12. If I change the code of this sketch and write the same commandos to send an sms as in the "Examples / examples from libraries / mkrNB / tools / send sms".  sketch, then the sms can be sent:    
  sms.beginSMS(senderNumber);
  sms.print(txtMsg);
  sms.endSMS();

CptHolzschnauz, 2021-07-22:
Do a firmware upgrade.

lavignema, 2021-09-01:
I also had issues with the Ublox modem not functioning properly. Initially the problems were numerous but an upgrade of the FW on the modem solved those issues. My remaining problem is that when I use the ReceiveSMS.ino sketch provided by Arduino my received texts are truncated at character #40 which is the first character of the time stamp. My sketch never actually sees the text body and therefore cannot react to instructions sent via SMS. I've documented this to Arduino technical support in emails back in February of this year.

CptHolzschnauz, 2021-09-02:
Try with dezimals (ascii encoding)
Example for "S":

```
 smspeek = sms.peek();
      Serial.println(smspeek);
      if (smspeek == 83)
      {Serial. print("it's a S!");}

```

lavignema, 2021-09-02:
Not sure what you mean Cpt.

I’m using the ReceiveSMS.ino library sketch provided by Arduino.

In it there is a command AT+CSCS? Which queries the modem for the character set in use.

It always returns +CSCS: “IRA” for the international reference alphabet. (default)

Are you saying I should forego using text mode and instead use PDU mode?

If so the library sketch will fail to parse the message as it is programmed for ascii and not hex values.

As I indicated in my post, the message I receive is printed to the serial interface without problem when initiated manually with AT+CMGL=”REC UNREAD”.

If I use the Arduino sketch I can see in debug mode the command AT+CMGL=”REC UNREAD” is sent to the modem and that the message is truncated after the 40th character prints to the serial interface.

Is this not what you are getting?

I sent screenshots to John Joyal at Arduino back in Feb. of this year and he concurred that he got the same results from the Arduino sketch.

 

Best regards,

MAL

lavignema, 2021-09-02:
> No, i had the same issue with the mkrnb lib after a firmware update.
>
> Keep IRA.
> Don't trust the serial monitor with the AT console(in general), sometimes you see not everything even when it's in the buffer.
> I think you want to identify a sms like they do in the sketch you mentioned.
>
> Give it a try. With the call it's:
>
>  if (sms.available())
> {

Yes, that would read the whole thing out but my goal is to get Arduino engineers to fix this sketch.

I still have to load the library in order to use SMS.read function so the memory payload is the same.

If I have to parse the messages myself like you say then I have additional code lines to inject.

Is it really that much extra code? No but…

My sketch is like 20,000 lines long right now and I am running out of programming memory.

I had this thing working in the 1400 but they discontinued support for 3G in the US so I had to port the code to another platform.

I had really hoped a friend of mine could swap the SARA-U201 for the newer SARA-R510M8S since they share the same fingerprint.

Unfortunately that was months ago and I haven’t heard him say he was successful.

Too bad Arduino doesn’t offer that modem on the board.

 

There’s another problem I’m seeing which is that after I receive a single SMS the modem fails to poll the provider for any more messages.

Sure the sketch sends the AT+CMGL=”REC UNREAD” command but the modem never actually receives it.

I had to use commands AT+CFUN=0, pause a few seconds, and send AT+CFUN=1 to reconnect.

After the reconnect I get another single SMS message.

I’m not sure if this is a bug in the modem or what.

If so then Ublox still has issues with their product and Arduino simply made a poor choice when they decided on the SARA-R410M.

 

Best regards,

MAL

When you want to read out the sms, use:

 
while ((c = sms.read()) != -1) {
          Received_SMS += char(c);
        }
Serial.print(Received_SMS);

CptHolzschnauz, 2021-09-02:
We are talking about MKRNB thus SARA410, right? With the 1400 it's complete different.
I had the same issue here with mkr1500 (just one sms download from the net per start) but this was fixed with the new modem firmware.
It's not a Arduino or lib issue, nobody will fix something because there is nothing to fix, it's the modem which is a seperate MCU you can only communicate to over a serial AT console. The AT commands from the lib are correct.
Ergo:
- Update to the latest modem firmware (You have to solder a USB cable to the MKR1500)
- Try again. It works here with me so it must somehow with you.

lavignema, 2021-09-02:
I beg to differ.

Yes I am talking about the NB1500 vs the GSM1400.

If you look at the SARA fingerprint on both boards they are identical.

If you look at the actual board schematic with Autodesk Eagle or similar, the differences in circuitry between the 1400 and the 1500 are minimal.

The actual differences are the omission of a few components, most notably the Schottky diode D3 in the charging circuit of the NB1500.

You can actually unsolder the Ublox modem from either board and replace it with one of the SARA-R500 series modems.

This is what I was referring to in my last email to you.

The R410M modem is strictly for use as NB/IoT whereas the R500 series are full blown cellular telephone modems with NB/IoT capability.

The R510M8S version even has a GPS built in so all you have to do is connect a GPS antenna to it and there is no need for a separate GPS board.

 

As far as the FW upgrade to the SARA modem you do not have to solder anything onto the board and I caution anyone not to do so.

There is a simple command that allows the modem to update over serial which is: AT+UFWUPD=3.

After the modem responds with: +UFWUPD: ONGOING you choose X-modem-1k transfer from file transfer menu.

Of course you cannot do this from the Arduino IDE but instead use ExtraPuTTy or even HyperTerminal.

I used ExtraPuTTy to do mine and my firmware is now at level: L0.0.00.00.05.08.A.02.04

This did solve the memory corruption issue and corrected multiple inconsistencies of operation.

It did not solve the single received SMS per connection issue.

That issue, I thought, would be addressed by Ublox but I have not seen another FW update for the R410M since.

 

I made John Joyal aware of my findings and emailed him a procedure to replicate my findings back in February.

He received them and did replicate my issue in their lab but have not updated the library since then.

Are you saying that the issue of only getting a single SMS text per connection is no longer an issue for you since your FW upgrade?

What version of FW are you running, and if different from what I’ve reported here, can you forward those files to me?

It is possible that you have a newer FW version that I have and it could solve all of my problems or only the SMS receive from the cellular provider issue.

Either way, I can always implement your fix and parse my own SMS messages if necessary.

 

Best regards,

MAL

CptHolzschnauz, 2021-09-02:
I beg to differ.

Yes I am talking about the NB1500 vs the GSM1400.

If you look at the SARA fingerprint on both boards they are identical.

If you look at the actual board schematic with Autodesk Eagle or similar, the differences in circuitry between the 1400 and the 1500 are minimal.

The actual differences are the omission of a few components, most notably the Schottky diode D3 in the charging circuit of the NB1500.

You can actually unsolder the Ublox modem from either board and replace it with one of the SARA-R500 series modems.

This is what I was referring to in my last email to you.

_Ok, got it cool. Have you done that with success?_

The R410M modem is strictly for use as NB/IoT whereas the R500 series are full blown cellular telephone modems with NB/IoT capability.

The R510M8S version even has a GPS built in so all you have to do is connect a GPS antenna to it and there is no need for a separate GPS board.

As far as the FW upgrade to the SARA modem you do not have to solder anything onto the board and I caution anyone not to do so.

There is a simple command that allows the modem to update over serial which is: AT+UFWUPD=3.

After the modem responds with: +UFWUPD: ONGOING you choose X-modem-1k transfer from file transfer menu.

Of course you cannot do this from the Arduino IDE but instead use ExtraPuTTy or even HyperTerminal.

I used ExtraPuTTy to do mine and my firmware is now at level: L0.0.00.00.05.08.A.02.04

This did solve the memory corruption issue and corrected multiple inconsistencies of operation.

It did not solve the single received SMS per connection issue.

That issue, I thought, would be addressed by Ublox but I have not seen another FW update for the R410M since.

_There is. 05.12_A2.19. Check out https://forum.arduino.cc/t/firmware-upgrade-for-ublox-sara-r410m-02b-on-the-mkr-nb-1500-2/699292/25
From .08 to .12 you have to solder the USB cable.

That update solved the sms issue.
_

I made John Joyal aware of my findings and emailed him a procedure to replicate my findings back in February.

He received them and did replicate my issue in their lab but have not updated the library since then.

Are you saying that the issue of only getting a single SMS text per connection is no longer an issue for you since your FW upgrade?

_.12 solved it_

What version of FW are you running, and if different from what I’ve reported here, can you forward those files to me?

_I think i'm not allowed. Check the post (see above) and you find it...._

It is possible that you have a newer FW version that I have and it could solve all of my problems or only the SMS receive from the cellular provider issue.

Either way, I can always implement your fix and parse my own SMS messages if necessary.

_Update the firmware, solves other problems with AT&T in the U.S._

Best regards,

Cpt.Holzschnauz

CptHolzschnauz, 2021-09-02:
PS: I suggested several time on https://forum.arduino.cc/t/mkr-1600-with-atsamd51-and-5g-alex-r5/898604 
to bring out a MKR1600 with a SAMD51 and a R500 or Alex modem. 

Soldering a R500 makes only sense when the I2S is connected, so only on a MKR1400.

Question: Have you ever soldered successfully a R500 on a MKR1400?

lavignema, 2021-09-02:
Yea, you’re right, they did disconnect the I2S on the NB1500.

It was just short of worthless on the GSM1400 since it was only one-way sound anyhow.

 

I just emailed my friend in Oregon who was going to attempt the swap for me.

Its been a pretty long time since I mailed him the parts too.

Not a good sign so I’m betting it didn’t work.

OTOH, no news is good news as they say, so I will keep my fingers crossed.

 

BTW, why doesn’t the ExtraPuTTy and serial update work for this FW rev?

Any info?

 

Best regards,

MAL

CptHolzschnauz, 2021-09-02:
I asked the same (i'm instrarep) and here's the answer:

https://forum.arduino.cc/t/firmware-upgrade-for-ublox-sara-r410m-02b-on-the-mkr-nb-1500-2/699292/4?u=intstarep

lavignema, 2021-09-03:
I noticed you left a link to the FW update file: https://xs.u-blox.com/sbha/SARA-R410M-02B-03-L0.0.00.00.05.12_A2.19_IP.zip

Unfortunately it is asking me for a userid and password.

Got one handy? 😉

 

Best regards,

MAL

 

https://forum.arduino.cc/t/firmware-upgrade-for-ublox-sara-r410m-02b-on-the-mkr-nb-1500-2/699292/4?u=intstarep

CptHolzschnauz, 2021-09-03:
Hint: You may say to a friend visiting you in your house: Be my … !
You … software to your computer from the internet.
I'm pretty sure you get it ;=)

lavignema, 2021-09-03:
I do get it and did get it.

Thanks Cpt.

CptHolzschnauz, 2021-09-04:
Cheers. Don't forget to close the issue.

lavignema, 2021-09-04:
Oh hell no. I’m pressing Arduino for factory support on this upgrade. I’d like to see an option for returning the device to Arduino for factory sanctioned FW update on this modem. I’m not comfortable with soldering on this pc board at all. I can do it but hesitate to alter on my own. Plus I want assurances that any NB1500 I purchase from here on will be fully updated with this FW rev. 

Regards,
Mike LaVigne

CptHolzschnauz, 2021-09-05:
DIY. Stay cool. It's easy. I'ts not a consumer device. What if you argue with Microsoft the same way? It's not worth the hassle.
Have a good one!

joshbober, 2021-09-07:
Chiming in because I have the same truncation issue that @lavignema reported above. My firmware is up to date (05.12), and I am able to receive the full SMS messages when I send AT commands through u-blox' app m-center (with SerialSARAPassthrough sketch loaded), but the ReceiveSMS.ino sketch fails for me in the same way.

It seems logical to me that this is a library issue; the modem is demonstrably able to receive and report received SMS messages using the +CMGL URC through m-center, so I know that messages are being received, stored and retrieved. I've been digging around the library trying to understand and seek a solution, but so far I have not found anything.

lavignema, 2021-09-07:
Josh,

 

I’m curious about the modem performance with FW rev 05.12

Can you tell me whether your modem now polls the cellular provider in a timely fashion for new SMS messages?

I have been dealing with a two-fold problem and to date.

One was seemingly Arduino based in their library as I related.

The other was clearly caused by Ublox firmware and concerned the receiving of SMS messages from the cellular provider. 

It appears that the modem will receive any messages the provider system has waiting when the initial connection is made.

After that it seems as though the modem simply fails to ask the cellular provider for new SMS messages.

I must use command AT+CFUN=0 to disconnect from the cellular provider after each SMS message I receive.

I then use command AT+CFUN=1 to reconnect and if there are SMS messages I will get them a single time.

I am not sure whether the modem will continue to poll until a single message is received or if it only retrieves messages already stored on the cellular system.

Does the modem now continuously receive SMS messages with the new FW rev?

 

Best regards,

MAL

joshbober, 2021-09-07:
MAL, 

Below is the AT terminal output after doing a little testing for you. I have added comments to explain what each step means. I sent messages to my NB1500 using 2 different phones with different intervals in between. The NB1500 is connected to the Ting network (AT&T). 

To directly answer one of your questions, my modem is receiving additional messages beyond the first one without any resets. To make this clear, I turned on the "New message indication" URC option. This sends a URC whenever a message is stored in the modem, without having to poll the modem.

This test was done using m-center, which I have had no problems with reading messages received- unlike ReceiveSMS.ino.

[16:41:45.116] AT+CMGF=1	// Set message format	
[16:41:45.131] OK
[16:41:45.131] 
[16:41:45.207] AT+CMGL="ALL"	// Refreshed message list
[16:41:45.222] OK
[16:41:45.222] 
[16:41:49.889] AT+CNMI=1,1		// Enable "new message indication" (URC that is sent when modem receives a new SMS)
[16:41:49.905] OK
[16:41:49.905] 
[16:41:49.981] AT+CNMI?
[16:41:49.997] +CNMI: 1,1,0,0,0
[16:41:49.997] OK
[16:41:49.997] 
[16:42:07.849] +CMTI: "ME",0	// URC: SMS received, stored in index 0
[16:42:31.224] AT+CMGF=1
[16:42:31.224] OK
[16:42:31.224]
[16:42:31.316] AT+CMGL="ALL"	// Refreshed message list, new message from phone 1 shown
[16:42:31.316] 
[16:42:31.331] +CMGL: 0,"REC UNREAD","+<<PHONE#1>>",,"21/09/07,13:42:07-28"
[16:42:31.331] Test 1
[16:42:31.331] 
[16:42:31.331] OK
[16:42:31.331] 
[16:42:40.974] +CMTI: "ME",1	// URC: SMS received, stored in index 1
[16:42:45.758] +CMTI: "ME",2	// URC: SMS received, stored in index 2
[16:42:45.774]
[16:42:48.460] AT+CMGF=1
[16:42:48.460] OK
[16:42:48.460]
[16:42:48.552] AT+CMGL="ALL"	// Refreshed message list, new messages from phone 2 shown
[16:42:48.552] 
[16:42:48.567] +CMGL: 0,"REC READ","+<<PHONE#1>>",,"21/09/07,13:42:07-28"
[16:42:48.567] Test 1
[16:42:48.567] +CMGL: 1,"REC UNREAD","+<<PHONE#2>>",,"21/09/07,13:42:40-28"
[16:42:48.582] Test 2
[16:42:48.582] +CMGL: 2,"REC UNREAD","+<<PHONE#2>>",,"21/09/07,13:42:45-28"
[16:42:48.582] Test 3
[16:42:48.582] 
[16:42:48.582] OK
[16:42:48.582]
[16:43:11.602] AT+CMGF=1
[16:43:11.602] OK
[16:43:11.602]
[16:43:11.694] AT+CMGL="ALL"	// Refreshed message list again, same messages stored shown as "read" now
[16:43:11.694] 
[16:43:11.694] +CMGL: 0,"REC READ","+<<PHONE#1>>",,"21/09/07,13:42:07-28"
[16:43:11.709] Test 1
[16:43:11.709] +CMGL: 1,"REC READ","+<<PHONE#2>>",,"21/09/07,13:42:40-28"
[16:43:11.709] Test 2
[16:43:11.709] +CMGL: 2,"REC READ","+<<PHONE#2>>",,"21/09/07,13:42:45-28"
[16:43:11.709] Test 3
[16:43:11.709] 
[16:43:11.709] OK
[16:43:11.709] 
[16:43:50.233] +CMTI: "ME",3	// URC: SMS received, stored in index 3
[16:43:50.248]
[16:43:53.735] AT+CMGF=1
[16:43:53.735] OK
[16:43:53.735]
[16:43:53.827] AT+CMGL="ALL"	// Refreshed message list again, new message from phone 1 shown
[16:43:53.827] 
[16:43:53.842] +CMGL: 0,"REC READ","+<<PHONE#1>>",,"21/09/07,13:42:07-28"
[16:43:53.842] Test 1
[16:43:53.842] +CMGL: 1,"REC READ","+<<PHONE#2>>",,"21/09/07,13:42:40-28"
[16:43:53.842] Test 2
[16:43:53.842] +CMGL: 2,"REC READ","+<<PHONE#2>>",,"21/09/07,13:42:45-28"
[16:43:53.858] Test 3
[16:43:53.858] +CMGL: 3,"REC UNREAD","+<<PHONE#1>>",,"21/09/07,13:43:48-28"
[16:43:53.858] Test 4
[16:43:53.858] 
[16:43:53.858] OK
[16:43:53.858]
[16:43:57.285] AT+CMGR=3	// Read message stored in index 3
[16:43:57.285] 
[16:43:57.285] +CMGR: "REC READ","+<<PHONE#1>>",,"21/09/07,13:43:48-28"
[16:43:57.285] Test 4
[16:43:57.285] 
[16:43:57.285] OK

lavignema, 2021-09-08:
Josh,

 

So that answers my question perfectly.

I can now assume that (with FW ver. 02.12)  I will no longer need to cycle the modem on/off line continuously.

 

Thanks for the time you took to examine this for me.

 

Best regards,

MAL

lavignema, 2021-09-09:
Cpt.,

 

While I appreciate your assessment of “easy” I gave it a try and found it to be difficult.

There is a fine line between attaching a wire to one of those two small USB pads and destroying the PC trace.

Unfortunately I fear I have now destroyed the trace as one of the two pads became dislodged from the board.

I can try to attach to the adjacent pad which is ¼ the size of the one I destroyed but I’m betting it will also be destroyed.

 

This can now only be corrected for me if Arduino is shipping replacement boards with the modem updated to 05.12 FW.

I have requested this information but received no reply as yet.

 

MAL

CptHolzschnauz, 2021-09-10:
Oh no, merde!  I guess you hadn't the rigth soldering iron? You need a narrow tip and then, most important, first of all bring some fresh solder or flux to the board, and after that with a presoldered cable holded at the place, a short shot of heat.
I hope that my instructions are understandable (Sorry, my english is not as it should be...). 
Anyway, fresh solder (flux) on both ends is most important! 
I think you gave to much heat because the old solder on the board did not melt right away? 

Never give up, mate!

lavignema, 2021-09-10:
Actually I use a small butane with a very fine tip. 
I’ve also been doing board level repairs for decades although lately I tend to use my hot air rework station instead of the butane. 
For whatever reason I opted for the butane this time. 
I pre-tinned all my wires and left a small blob of solder hanging at the tip of each wire.
As you said, I just wanted to momentarily touch the iron to the board to fuse the connections. 
For whatever reason, possibly a defective pad, the wire and pad detached shortly after the process.
I know the procedure was a valid one because one wire is still securely attached to the other USB pad.  
Like I said, if I can get assurances from Arduino that all new boards are fully updated to 05.12 FW then I’ll just replace it with new. 
I have one more of these in my shop and if Arduino provides a return for upgrade at a reasonable price I will take advantage of that rather than chance another toasted PCB. 
Please don’t feel you should apologize for your English. 
I’m sure If I were to attempt to communicate in your first language it would be atrocious. 

Regards,
Mike LaVigne

CptHolzschnauz, 2021-09-10:
;=)
I don't think you will get one with 5.12 out of the factory, they just solder the modem (delivered to them some time ago)to the board and that's it.
I already updated (soldered the USB cable) to four mkr1500, mostly with windows problems (as mentioned in the arduino forum) but with some help from above i made all of them. So if it's possible for me, it would be for you. 
To be fair, the issuer of the problem is u-blox with their software (politics) and not arduino.
I blame arduino that they didn't connected the I2S pin because voice is possible with LTE-M1

Best regards

lavignema, 2021-09-10:
I think it was clear from the original design that Arduino did not want anyone making their own cell phone out of them. Maybe to avoid legal hassles. Ublox clearly created this monster. It wouldn’t surprise me one bit to find that the SMS library function is still in conflict with the modem firmware. I’ve examined the library and can see no error in programming. It parses the messages very effectively and can easily be modified by the user to perform many additional functions. We shall see.

Best regards,
Mike LaVigne

;=)

CptHolzschnauz, 2021-09-20:
Downgrade the SAMD21 core to 1.8.9 or 1.6.21 and try again. There is a problem with the buffer with MODEM in 1.8.11.
Best regards!

lavignema, 2021-09-20:
Thanks Cpt.
I’m still going to try and update that modem FW. I plan to hold the conductor onto the remaining pad with a straight pin. After all, it’s only this one time and after that Ublox says it will go in over FOAT/FOTA.

Br,
MAL

Best regards,
Mike LaVigne

lavignema, 2021-09-21:
I wonder if I might ask for a second opinion on my NB1500 board. I managed to use a straight pin to make contact on the one damaged USB pin and when I plug the modem into a port I get the “USB device connected” sound from my laptop. The problem now seems to be that the device is unrecognized by Win10 and in Device Manager under USB devices it says “Unknown USB Device (Device Descriptor Request Failed)”. I tried plugging it into another computer here and it also reports the same. This tends to make me believe that the modem might have somehow gotten damaged. Is it possible that I simply do not have the drivers and this could be the result? I have not been able to confirm that the Qualcomm drivers called for in the posts are in fact installed.

Any Insight is appreciated.

 

Bet regards,

MAL

CptHolzschnauz, 2021-09-21:
You have to power up the mkr over the "normal" USB, then connect the modem. If the Qualcomm driver is instelled it can be a weak cable, i had the same problem a week ago, a well looking USB cable was made connection, the USB device was recognized  but the computer was not able to recognize what kind of device. Exchanged the cable - everything was fine.
Ergo: You can't tell from the outer look of a USB cable if it's trash.

lavignema, 2021-09-21:
Thanks again Cpt. ,
I’m pretty sure the cable I scavenged was very old. I’m betting this modem requires USB 2.0 or3.0 even. 
BR,
MAL

Regards,
Mike LaVigne

lavignema, 2021-09-21:
BTW, I’m not sure what source to find EasyFlash or Qualcomm drivers. Any suggestions?

Regards,
Mike LaVigne

CptHolzschnauz, 2021-09-21:
https://forum.arduino.cc/t/firmware-upgrade-for-ublox-sara-r410m-02b-on-the-mkr-nb-1500-2/699292/15?u=intstarep

lavignema, 2021-09-29:
Hey Cpt.,

I wish to report that I successfully upgraded one of two NB1500 units.

It was indeed the cable quality as you surmised.

I found one that had been a communications cable for an old cell phone.

When I cut the phone connector off I could see major differences from the first one I scavenged.

This one had a definite twist to the conductors as well as an inner drain, foil shield, copper braid over the foil and a separate outer drain.

Anyway it connected at first but the USB was on and then off again every few seconds.

I loaded the Arduino with the passthrough sketch with the on/off button set to DI6.

I toggled the on/off state and the modem then connected and stayed connected.

I tested the communications with m-center which was able to communicate at 115200.

I next ran the Easy Flash 12.10 program and voila, it connected immediately, and updated the FW package.

I now have my modem reporting the latest FW version via ATI9 command in the standard Arduino SerialSARAPassthrough.ino sketch.

Thanks for your help.

Now I will try to patch this new cable onto the wires I’ve soldered to the original NB1500 and try once more to achieve this feat.

I am optimistic.

 

Best regards,

MAL

 

https://forum.arduino.cc/t/firmware-upgrade-for-ublox-sara-r410m-02b-on-the-mkr-nb-1500-2/699292/15?u=intstarep

CptHolzschnauz, 2021-09-29:
Hi MAL
Good to hear. A weak USB cable can fool you a lot, i know best. Next time you don't need to solder and perform that DI6 switch thing. The difficult to solder connection near the pins and the GND are also absolete because this is just V+ and GND which can be connected via the normal MKR pins. So just the two big easy to solder connections are neccesary.
Have a good one!

lavignema, 2021-09-30:
Hey there Cpt.,

 

I must be way more talented than I ever dreamed because I just updated the board I thought I had destroyed.

When a second conductor dislodged from the board (the +5v connection to the USB enable pin of the modem) I thought I was totally hosed.

But as luck would have it, I put a hat pin on one end of an alligator jumper and attached the other to the +5V red wire.

I soldered an old 1/8 watt resistor to the green (USB) wire and clipped the extra lead so as to fashion a sort of pinpoint probe.

Using the schematic and Eagle software I isolated the PC trace where it went through to the modem enable pin.

I poked the hat pin into the board to pierce the coating and then held the green wire onto the little dot adjacent to where the first pad dislodged.

I heard the computer give that USB connect sound and watched as the device appeared in my device manager screen with no indication of malfunction.

I had to hold both connections steady with one hand and activate the Easy Flash program with the other but I managed to do just that.

I kept telling myself that I only had to hold this for a minute as I watched the download proceed from one step to the next.

I managed to stay steady for that entire minute and then I heard the beep indicating the update had succeeded.

I immediately unsoldered the wires and loaded the SerialSARAPassthrough.ino sketch to run ATI9 and it indicated the new FW version was present.

Again, Cpt. Thanks for all your help.

 

I’ve already told Arduino that I cannot purchase any more product until they can assure me the latest FW version is on the modem.

I’ve also ran my SAMD software version back to 1.8.9 as you advised.

I recall not having any problem with receiving an SMS using the provided library functions with that version.

Hopefully Arduino will catch their buffer problem and issue a new update that addresses it.

 

Best regards,

MAL

lavignema, 2021-10-03:
Hey, Cpt.,

 

I wonder if I might impose upon you and/or the entire world for that matter regarding my NB1500 functionality.

I recall when I first got the device that I had difficulty getting it to send or receive SMS text messages.

I performed the 1st FW update using the Extra Putty program and uploading it to the device with Xmodem I think.

Once the update was complete I then found I could send and receive SMS texts but it was spotty.

I now have updated the FW for the 2nd time and the device no longer will send or receive SMS texts.

As a test, I placed the (Verizon) SIM I am using into my cell phone and was able to send and receive SMS texts.

So I now must conclude that there is some sort of configuration command that the cell phone performs that I am missing.

Is there a simple command that must be given at the time of connecting to the cell provider that will enable the SMS text system?

I am actually not using the device for its IOT capability but rather for now at least using the SMS text system to relay data and commands.

Is there a special setup to use the SARA-R410M as a simple SMS text interface?

When I connect to Verizon I get the +CEREG=0,1 and +CGDCONT+1,”IP”,”VZWINTERNET”, …all the tcp/ip info, so it seems proper.

 

Best regards,

MAL

CptHolzschnauz, 2021-10-04:
The whole world is listening! 
Have you tried to send/receive a sms with the windows software from u-blox called m-center.exe?

lavignema, 2021-10-05:
No, I have not. Thanks for the suggestion.

 

Best regards,

MAL

CptHolzschnauz, 2021-10-19:
Puhh, i know that there are Verizon special cases coded in the firmware but i have no experience nor testing possibilities so i can't really help in this case. Check the u-blox AT-CommandsExamples_AppNote_(UBX-13001820).pdf from page 59 on.

TomWesthoff, 2023-08-25:
FYI - I found out recently that Verizon blocks incoming SMS from +882 numbers. AT&T and T-Mobile do not block. I am using Hologram.io SIM and their outgoing SMS come from +882 numbers. Trying to find another SIM provider who allows SMS and doesn't send SMS from an International Network.

> International Networks is the name given by the [International Telecommunication Union](https://en.m.wikipedia.org/wiki/International_Telecommunication_Union) (ITU) to [country calling codes](https://en.m.wikipedia.org/wiki/List_of_country_calling_codes) +882 and +883, and serves as a catch-all for telephone services not dedicated to a single country. [Satellite telephone](https://en.m.wikipedia.org/wiki/Satellite_telephone) carriers, especially those with worldwide service, are allocated within the [Global Mobile Satellite System](https://en.m.wikipedia.org/wiki/Global_Mobile_Satellite_System) (GMSS), country code +881, with the exception of non-terrestrial [Inmarsat](https://en.m.wikipedia.org/wiki/Inmarsat), country code 870.

lavignema, 2023-08-25:
Hello Tom,

 

I’ve been trying to get my NB1500 to function on the SMS system for several years now with no success.

I’d love to see what your solution will do for my setup but I’ve already tried T-Mobile without the results I’d hoped for.

I began with a lengthy program that functioned properly using the original GSM1400 but the 3G support evaporated and was left facing migration.

Then I faced the FW issues of the new modem on the NB1500 and eventually got mine updated no thanks to UBLOX.

Once I achieved connection to the cellular system, I then found that none of the SMS functions would work.

My system uses the SMS system to update the user on operational status of the device as well as function programming.

Can I impose on you to provide the startup code you use to initialize the modem and send SMS messages successfully?

 

Regards,

Mike LaVigne

+1(804)240-2980

TomWesthoff, 2023-08-25:
I can try to help, but looking at previous posts, you are way ahead of me, I think. I did recently have a good interaction with U-Blox support and am still sorting through what I learned and trying to come up with a good initialization sequence.
What brand SIM are you using? Are you having problems sending or receiving SMS, or both? 
What modem firmware are you using? I'm at ..05.11, A.02.16.  
I'm trying to get to the newer firmware but I'm confused as to which of the many methods talked about above is the easiest way to do it. Maybe you could help me with that?

Sorry for all of the questions, I just want to understand what your setup is. Happy to help if I can.

lavignema, 2023-08-30:
Tom,

 

Possibly I just need to update my Arduino board to the latest version along with the UBLOX firmware.

Were you able to update your firmware without attaching anything to the NB1500 board?

Last update I performed; they (UBLOX) swore to never again require separate comms channel into the modem.

Last update I performed was A.02.04

Would you happen to have the zip file you performed your update with?

Also, what was your original FW version before ending up with A.02.16?

The files I used for my update were named L0506A0200-to-L0508A0204.zip and L0506A0201-to-L0508A0204.zip.

You had to do ATI9 to get the present firmware and apply the appropriate file for the update.

 

Regards,

Mike LaVigne

(804)240-2980