arduino-libraries/MKRNB #57: MKRNB receiving SMS halts after a short time

CptHolzschnauz, 2020-08-14:
Hi there

After the ugly hardware problems with the MKR1400 (Switching to 2G freezes the code) i swaped to the MKR1500 facing the next frustrating problem..

The board is receiving sms right after startup but then AT+CMGL="REC UNREAD" on and on.
After a reset the SMS come in, one after the other, like they had waited somewhere in a memory...

I'm testing with the "official" example from the MKRNB, the ReceiveSMS sketch by Javier Zorzano.
I already updated the Modem Firmware 
ATI9
L0.0.00.00.05.08,A.02.04

ATI
Manufacturer: u-blox
Model: SARA-R410M-02B
Revision: L0.0.00.00.05.08 [Apr 17 2019 19:34:02]
SVN: 02
MEI: 352753090827732

with no remedy...

Any help or hints or additional questions?
Thx

janakelarsson, 2020-08-15:
You are sure you didn't write "REC READ" via passthrough? You can only read each SMS once using the library. It immediately disappears from the "REC UNREAD" list, and is instead accessible as "REC READ". The library is not written to access those.

CptHolzschnauz, 2020-08-16:
Yes, i was, but now i cant't reproduce but still can't understand whats happening now..
1. The sketch does not receive the sms
Endless looping 
(AT+CMGL="REC UNREAD"
OK)
2. I change to passtrough and the SMS is marked as read. Why? Who has read it?

This night somethin strange happend: I sent yesterday a sms that hangs again and i walked away for a beer and let the sketch run. After hours, in the middle of the night the sms went trough and i get the answer...

Ergo: The first 30 seconds in the sketch everything works, then nothing or hours to receive. It's frustrating hunting flaky bugs...

janakelarsson, 2020-08-16:
You read it. Each SMS is only readable once.

The reason for the "strange" thing is that SMS itself is not reliable. The delivery is not guaranteed. There can be a long delay before it is delivered.

CptHolzschnauz, 2020-08-16:
Yes, i do understand but the SMS are stored in modems memory immediatly. Then, for a strange kind of reason marked as read without passing the content in return of sms.avaiable();

What you say about delivery is right but very theoretical, i think sms is delivered immediatly in 99.999998% of the cases, aren't they?

janakelarsson, 2020-08-16:
I'd say 98-99%. It does happen to me from time to time that SMS are late. Perhaps in Switzerland you have better network coverage?

janakelarsson, 2020-08-16:
Also, I use my fork of the library, the SMS handling code is rewritten there.

CptHolzschnauz, 2020-08-16:
Things Mobile has contracts with different swiss carriers, i can't compare much.

I compiled again with your fork - same behavior. It proceed the SMS after a hardware reset.

CptHolzschnauz, 2020-08-16:
After several hours spent i must correct myself. It's the modem, not the library. The SARA Module grabs the SMS from the network only after a reboot. It's not the lib, the sketch or the network operator. It's the modem, despite the new firmware.
I found no way to trigger the sms receiving, I'm out of solitude. Anyway, this is a task for u-blox.

janakelarsson, 2020-08-16:
That is my experience too. In my case, since I eventually bricked the modem, I have replaced it and then everything works. 

But the first thing I did was to update the firmware. I read somewhere that the new firmware is more resilient against unexpected power-downs. 

If you already damaged some setting with the old firmware, I do not know how to reverse the damage. It is possible that upgrading the firmware does not help.

CptHolzschnauz, 2020-08-16:
It seems to be. The modem is f.u.
But how is it possible to brick a hardware with software when i still can reach the hardware with X-modem-1k and replace the firmware? Arghh
Looks like fanbase(x-blox)= fanbase(x-blox)-2
Thx anyway

CptHolzschnauz, 2020-08-20:
I did. No success. Same behaviour. New MK1500 out of the box, new firmware first, standard sketch (Sms receive), same issue, even larger frustration... Receiving an sms in the year of 2020....

janakelarsson, 2020-08-20:
Which MNO profile? Are you registered for both data and SMS/voice? Did you turn off power-save mode?

CptHolzschnauz, 2020-08-20:
Yes, double checked. Followed your path 

    Select network operator "standard Europe": AT+UMNOPROF=100
    Reboot: AT+CFUN=15
    Issue AT until you get OK so the modem is rebooted
    Then quickly turn off powersave, otherwise modem turns off in seconds: AT+CPSMS=0
    Select combined attach, both speech/SMS and data: AT+USVCDOMAIN=2

But i did a new record! I reveived 3 (three) sms before ending in the loop....AT+CMGL="REC UNREAD" OK!

I wrote also a ticket to u-blox

janakelarsson, 2020-08-20:
The code I have checks if the modem is registered on the network before checking available SMS, in the preamble use

`NBPIN             reg;`

And then

`  if (reg.checkReg()==0) {
    while (sms.available()) { ...`

The else clause restarts the modem.

CptHolzschnauz, 2020-08-20:
And still looping....
14:00:09.616 -> AT+CMGL="REC UNREAD"
14:00:09.616 -> OK
14:00:10.613 -> checkReg:
14:00:10.613 -> AT+CREG?
14:00:10.613 -> +CREG: 0,5
14:00:10.613 -> OK

How do you restart the modem properly? With AT+CFUN=15 ?
Desperately seeking for answers... AT+CREG 0,7 (Sms Only, Roaming) ? Other strange ideas?
My main question is : How is it possible that everything works perfect  for about 30 seconds, up to 3 sms and then stuck?
It's not the SIM, it's not the carrier, with every mobile phone everything works perfect...

janakelarsson, 2020-08-20:
+CREG: 0,5 tells you that you ARE registered, so that is not the error.

(I restart with the library `if (gsm.begin(PIN)==NB_READY) {...`)

I don't have more ideas now.

CptHolzschnauz, 2020-08-20:
I know, that's why my frustration level grows...
Thank you anyway for your help

janakelarsson, 2020-08-20:
ISTR I had accidentally filled some receive buffer. What does AT+CMGL="ALL" return? 

(I added a `.clear()` method to the library to clear the whole SMS memory of the modem.)

CptHolzschnauz, 2020-08-20:
Nothing. The modem is not catching the SMS from the network.

CptHolzschnauz, 2020-08-21:
After other tryings i can say, it's just about running time. After startup, the modem (and the library)is working with unlimited sms, after about 20 to 30 seconds the modem goes  endless
AT+CMGL="REC UNREAD"
OK
And does not loading down the sms hanging in the mobile network.

CptHolzschnauz, 2020-08-26:
Mr. janakelarsson, you wrote
> (I added a `.clear()` method to the library to clear the whole SMS memory of the modem.)

In that method you use   MODEM.sendf("AT+CMGD=0,2");

Shouldn't it be 1,1 or 0,4? 
Don't let 0,2 leave the unread SMS untouched? 
I just discovered that my modem received around 120 unread ad sms and corona warning sms from the governement. Because the were never read they remain in the memory which fills it with garbage.
May it be a good idea to delete all (also unread) during the setup?
I do now : 
Serial.print("SMS memory flush");
MODEM.sendf("AT+CMGD=0,4");
MODEM.sendf("AT+CMGD=1,4");

In the uBlox AT example manual is written:
AT+CMGD=1,1
Delete all the read messages, leaving unread and stored (whether sent or not) messages untouched
Which worked to my surprise, this command deleted all the read sms, i expected just a delete of the sms in the index 1
so in my strange logic
MODEM.sendf("AT+CMGD=1,4"); will delete all sms?

janakelarsson, 2020-08-26:
The +CMGD command takes two arguments, one mandatory `index` and one optional `flag`. If `flag` is present, then `index` is ignored.

I chose `AT+CMGD=0,2` because then unsent and unread SMS are not deleted. (`AT+CMGD=0,1` preserves sent, unsent, and unread messages, while `AT+CMGD=0,3` only preserves unread messages, and finally `AT+CMGD=0,4` preserves no messages.)

I kept unsent messages because these will be sent on registration to the network, and unread messages because these will be read by your code (I presume), so both can be deleted by a later call to `.clear()`.

CptHolzschnauz, 2020-08-26:
Thx a bunch ! I do MODEM.sendf("AT+CMGD=0,4"); in setup to clear the canvas...

CptHolzschnauz, 2020-08-31:
After a helpful hint that the u-blox m-connect software can be connected trough serial passtrough (So easy and i didn't know that...)
i reproduced the same behaviour with their own software.
I send some sms to the modem from my cellular, first one came in, the following SMS hang in the network. Then, AFTER SENDING A SMS from the modem via m-connect, the sms were downloaded and shown. Same behavior with the Arduino library. So I really think it’s a bug in their firmware, not a issue with the MKRNB Lib.

janakelarsson, 2020-08-31:
I suspect that the weird behavior was in my case caused by an unexpected shutdown from using the RESET_N pin too liberally. The firmware should in my opinion be written so that it is secure against such events.

CptHolzschnauz, 2020-09-06:
@janakelarsson : I don't think so, i have the behaviour with a brand new gear. One question:
You are now able to receive as many sms as you want without sending one with the MKR1500? 

@all
I spent another hours on debugging. I can reproduce the bug with 3 different operators with 3 different SIM cards on the u-blox software m-center. The only explanation i see is a u-blox firmware bug.

The first 1 to 3 sms come in like they should, after that, only a sended sms deblock the waiting sms in the network. Then I receive one sms until I send another one. Same issue.

U-blox wrote:
"I was unable to reproduce this issue with my SARA-R4 EVK. I made multiple tests and received/sent all the SMSs, and got updated everytime. I have requested our AEs/R&D if they can reproduce this. We are working on it. I will update you on this. "

I can’t imagine nobody is able to reproduce this error. I am at the end, dazed and confused.

janakelarsson, 2020-09-07:
Yes, the current sketch receives an SMS command, and responds via SMS. There can be a long delay until next command.

CptHolzschnauz, 2020-09-11:
There is a new firmware for the modem but my issue is still not solved..maybe others.

janakelarsson, 2020-09-12:
Changelog?

CptHolzschnauz, 2020-09-13:
Firmware is closed source. My personal changelog is : Even worse...
With the new firmware i am only able to receive a sms AFTER i sended one. The behaviour is still the same. Incoming sms hangs in the network until i send one, then they drop in. The technician from ublox can not reproduce the error, he sended me his AT protocol, i sended him my protocol, mine was made with the u-center software from u-blox. Something is rotten in the state of denmark.. I can't solve the issue, the problem is between the network and the modem thus not in my reach.

CptHolzschnauz, 2021-04-04:
I can confirm with the new firmware L0.0.00.00.05.12,A.02.19 this problem is solved. 
Instructions for the update under: https://forum.arduino.cc/index.php?topic=731275.0

closed 2021-04-04 by CptHolzschnauz