arduino-libraries/MKRNB #58: Missing AmazonRootCA1 ?

lakidd, 2020-08-18:
Hi 
I see that this commit 
https://github.com/arduino-libraries/MKRNB/commit/7891975dff5fb697d0115670ea4ed19b9f3695ab
removed the AmazonRootCA1 from the NBRootCerts.h file. 
Will that break connections to AWS IoT if the MKR NB 1500 trys to connect to AWS IoT core over SSL ?
I'm trying to debug one at the moment (logs not available, will have them in the next few days) and it's not connection to AWS IoT core, the MQTT connection is not coming up.
Why was the Amazon root cert removed ?

torntrousers, 2022-03-21:
I can't get the MKRNB1500 to connect to AWS IoT, so this is curious. What's the story of that commit @sandeepmistry ?

marcelchelo, 2022-10-11:
I also can't get mar 1500 to connect to AWS iot.
However I was successful with azure.