arduino-libraries/MKRNB #59: Failed MQTT Connection Error Code 3

nico-fri, 2020-11-12:
Hi,
i tried to send data with MKR NB to a MQTT Broker with following components:
- MKR NB 1500
- 1NCE card with active data plan
and used MKRNB and ArduinoMqttClient Librarys.
When I try to connect the MKR NB to the MQTT broker, I get ERROR-CODE 3 = Server unavailable. With the MKR 1010 Wifi the connection to the broker always works. Even with the program MQTT.fx and the same access data, the connection to the broker works without problems and data can be sent. The connection with the MKR NB has already worked and now it gives the error without changes in the code again. When I run the NBWebClient example sketch, I get the http request without any problems.

Any help or hints or additional questions?
Thanks

ircole, 2020-11-13:
I'm seeing that error intermittently.  I'm using the soft watchdog code 
to reset the processor when there is some hang and sometimes it will 
recover but in some cases I have to remove the battery (or if no battery 
the USB cable) to get it to work again.  A few days ago I picked up the 
pre-release of the MKRNB library and before that it seems that the reset 
by the soft watchdog always recovered.  Don't know if the problem is 
with the update library or something else. It's not uncommon to get this 
error as it's caused by a timeout on a write.  Setting the timeout in 
NBClient write to 240000 helps but still get problems just not as 
frequently.  Seems after a timeout on a send often the library code 
hangs in some loop and my code never gets control again which is the 
reason I use the software watchdog. My broker is a Mosquitto broker 
running on one of my systems and I can see in the log that the open 
socket works as the broker logs a New Connection message.  Before 
updating the library I could run for days (with many watchdog restarts) 
but now I can't.  Unfortunately I don't have a solution.  At this very 
point in time I'm having success send messages to the broker but it quit 
earlier today and it took completely removing power from the 1500 to get 
it working again.

AT+USOCR=6
+USOCR: 0
OK
AT+USOCO=0,"207.40.144.108",1883
OK
AT+USOWR=0,56,"103600044D51545404C2003C00174D4B52313530302D333532373533303930383630383038000773656E736F72730008747269636B68616D"NCClient 
send wait returns: -1
AT+USOCL=0
ERROR
.CONNECT ERROR: 3

+UUSOCL: 0

SahithyaYadav, 2021-07-29:
Hi,
I am registering MKRGSM 1400 with Azure IoT Hub. The device registered successfully, after registering the device, when I am connecting the board to Azure IoT Hub, I am getting error -3(Connection refused: server unavailable). But after few hours(around 12 hours) i could able to communicate with Azure IoT Hub without any error.

+USORD: 0,91,"1403030001011603030050AB9F56C7E6C09842F3ECE857D9C77604C549C6E823AB6411918872C9864A0036EFB2A3E965CF2920D522500B109E29DCCDACA7D411E8F04723B50281522CC821533F28A4AF1ABF0A0359648CC35832D0"

OK
AT+USORD=0,512

+USORD: 0,0,""

OK
AT+USOWR=0,149,"1703030090532471CABDDE6818D1D2DDEE8AD8D785AFA7BE926AC9CEB5063F8D8A8082FCD5C6054DD278F1BB19FF4BBAE43781A180B6DC681C3910E1CCF7C9F08D72B583BD028AF6AEC84405F652AF28F8D8EF23B41A971B189B1DA235E5AD6F6435107862D5D63CBF107D95B34774FDDC962FE98A5DA468805C139200C29F3A1193305C92087A4F14026FA4EA76E6B451602FB321"
ERROR

+UUSOCL: 0
AT+USOCL=0
ERROR
.3
AT+USOCR=6
+USOCR: 0

@ircole , if u resolve your issue, It would be very helpful if you suggest any. Thanks.

CptHolzschnauz, 2021-08-22:
try https://github.com/arduino-libraries/MKRNB/pull/84