arduino-libraries/MKRNB #71: nbAccess.begin fails on second loop of gprs.test example sketch

calvino80, 2021-01-14:
arduino model: MKR NB 1500
modem model: SARA-R410M rev. 02B-00
modem firmware: 05.08
MKRNB library version: 1.5.1

with the example sketch gprs_test, the modem will not complete more than one loop unless line 58 is changed from these values:

if (nbAccess.begin("") != NB_READY) {
....

to these values: 

if (nbAccess.begin("", false, true) != NB_READY) {  
....

Mike

giulcioffi, 2021-02-09:
Hi @calvino80, could you please test the changes that I proposed in PR #72 and let me know if this solves your issue?

joshbober, 2021-07-12:
Weighing in here with results from my testing:

I think that the "hanging" is a result of the code entering a "while(true);" statement in case of failure to connect. 

```
if (nbAccess.begin(PINNUMBER,"wireless.dish.com") != NB_READY) {
  Serial.println(errortext);
  while (true);
}
Serial.println(oktext);
```

I am currently diagnosing issues with connection to my provider, so running the example sketch with this code always results in the error condition and loop hanging. I do not know if this is also the case for @calvino80, but if so, I think the real issue is with the example sketch as coded.

My recommendation is to use the connection loop from the SendSMS example sketch (or similar). If the modem needs to be restarted in between "failure-to-connects", then simply set a boolean "restart" variable that gets passed in on subsequent connection attempts. Example below:
```
  Serial.print("Connecting NB IoT / LTE Cat M1 network...");
  bool connected = false;
  bool restart = false;
  while (!connected) {
    if (nbAccess.begin(PINNUMBER,"",restart,true) == NB_READY) {
      connected = true;
    } else {
      Serial.println("Error - could not connect");
      restart = true;
      delay(1000);
    }
  }
  Serial.println(oktext);
```

PR #72 causes a modem restart on first execution of the NB begin(). I do not know if this is necessary, since (at least in my case), the first execution of begin() occurs at startup, so it may be considered redundant unless I am misunderstanding something.