arduino-libraries/MKRNB #79: Unpredictable +CME ERROR after hours of AT+USOWR

CptHolzschnauz, 2021-07-20:
Hi

I'm able to send data over a socket with AT+USOWR for hours and hours and out of the blue i get a 
+CME ERROR: Operation not allowed
and after that, the modem is totally paralized and does not react at all anymore, only a hardware reset brings it back to live.
Reproduced on 3 different MKRs,  all within the timeouts, the other side does not disconnect, the payload is ok, the buffer is correct (The  AT+USOWR is formated right). As i said before, the same command works well for hours.
After the failure, the modems Internal TCP/UDP/IP stack class error codes (Get Socket Error +USOER)
is 65 EEOF - End of file.
- I know bugs like these are very hard to track down, any hints?
- Wouldn't it be nice to have an error handler to handle this? 

Logs:
For hours:
> AT+USOWR=1,310,"Here are 620 bytes, i just delete  for better readability"
 -> +USOWR: 1,310
 -> 
-> OK

And then, sometimes after minutes, sometimes after hours:
19:31:14.010 -> AT+USOWR=1,310,"Here are 620 bytes, i just delete  for better readability"
19:31:20.182 -> +CME ERROR: Operation not allowed
and the modem does not answer anymore until a hard reset.

zbelding, 2021-07-21:
> And then, sometimes after minutes, sometimes after hours:
> 19:31:14.010 -> AT+USOWR=1,310,"Here are 620 bytes, i just delete for better readability"
> 19:31:20.182 -> +CME ERROR: Operation not allowed
> and the modem does not answer anymore until a hard reset.

So the modem locks up **immediately** after replying +CME ERROR?

CptHolzschnauz, 2021-07-21:
Dear zbelding
More or less, depends on the definition of immediately. See first a normal operation:

12:36:22.367 -> AT+USOWR=0,256,"PAYLOAD Data"
12:36:22.600 -> +USOWR: 0,256
12:36:22.600 -> OK
12:36:22.642 -> +UUSORD: 0,1860
12:36:22.642 -> AT+USOWR=0,54,"PAYLOAD DATA"
12:36:22.642 -> +USOWR: 0,54
12:36:22.642 -> OK

Here's the accident:

12:37:28.351 -> AT+USOWR=0,256,"PAYLOAD DATA"
12:37:41.463 -> +CME ERROR: Operation not allowed
12:37:41.463 -> 
12:37:41.463 -> 
12:37:41.463 -> +UUSORD: 0,2170
Then is flatline.
So a long time between write into the socket and the error msg (Because the socket is dead, but why?) and a short period between the error msg and flatline.

I did some tests with other mqtt brokers, but with all three (Official Mosquitto, self hosted mosquitto and shiftr.io) the error occurs at least after around 6 hours. On the server side i never found any problems.

To be honest, i don't think it's a lib bug, i think  it's the f.... Sara Modem...or the network.
With all the adjustments, the profiles, the MQTT mode, the blablabla but then you have to fight with simple things like sms and a tcp socket....gdmmt...
Have you found a way to do a factory reset? Because during the FW update its safes the NV back to the modem...while fighting a long war i found out that diffenrent mkr1500's reacts also different...

Anyway, the lib must handle these errors somehow....

Thx. A fan of yours

PS BTW, with zbelding on the thread, have you ever saw that one?:
https://www.telit.com/developer-zone/charlie/
A MKR1500 Turbocharged without libs, a seperate usb for the modem, gps and acceleration sensor on board ...maybe a remedy for us ;=) ? Or your first modem lib?  ;=)

CptHolzschnauz, 2021-07-21:
PS 2: At the moment i'm facing out of the blue another problem with receiving sms....
https://github.com/arduino-libraries/MKRNB/issues/80
Keep smiling

zbelding, 2021-07-21:
> 12:36:22.642 -> +UUSORD: 0,1860

This +UUSORD is the modem telling you that it has buffered a response from the server. Are you reading this response? It can take a little bit of time for this response to come in. Trying to send more data before the last response has been fully buffered could be why it is giving you an error. Also, the SARA buffer can only hold so much data. Reading the data will clear the buffer. Also closing the socket will clear the buffer(USOCL).

Sorry, I wont be any help with the SMS. I use an http API(clickatell) to send messages. I don't have any experience with built-in SMS messaging.

CptHolzschnauz, 2021-07-21:
Strike! While i was fixated on writing to the socket the read buffer is full...the modem tells me but i was blinded by anger and despair... 
Thx for the hint!!

CptHolzschnauz, 2021-07-21:
Nope, i closed it too early.

22:44:38.730 -> AT+USORD=0,512
22:44:38.730 -> +USORD: 0,""
22:44:38.730 -> 
22:44:38.730 -> OK
22:44:38.798 -> AT+USOWR=0,256,"PAYLOAD DATA"
22:44:52.128 -> +CME ERROR: Operation not allowed

zbelding, 2021-07-22:
Are you sure the modem is locked up, or is it just the library not trying again after receiving error? When get an error like this, I usually close the socket, back up and try again. It usually works on second try.

CptHolzschnauz, 2021-07-22:
The good old MS way to handle the problems ;=) Restart! 
No, you sent my on the right path, it seems that the lib is not reading/freeing the buffers properly. I'm workin (fightin) on it.
(The CMEE=2 doesn't make the modem to verbose too much...so i'm groping around in the dark)
Thx a lot!

closed 2021-07-22 by CptHolzschnauz

CptHolzschnauz, 2021-07-23:
If someone found this topic on the hunt for a bug:
My problem seems to come from a timer function. To post a MQTT message every minute i set up a "timer":

 static unsigned long delayStarttime;
    unsigned long currentTime = millis();
    if (currentTime - delayStarttime >= 60000UL)
{mqtt message}

In the loop() i also made communication with the modem, to check for signal quality:
signal_strenght = scannerNetworks.getSignalStrength();

So the loop has different duration (+- some seconds) and so sooner or later the AT commands to the modem colides.
Solution:
I do everything related to the modem inside a timer:

static unsigned long delayStarttime;
unsigned long currentTime = millis();
    if (currentTime - delayStarttime >= 15000UL)
{get signal strenght}
    if (currentTime - delayStarttime >= 30000UL)
{do a mqtt client loop}
 if (currentTime - delayStarttime >= 60000UL)
{do a mqtt push, if it's the last one :  delayStarttime = currentTime;}

So the modem has time to digest the AT commands and they come one by one.

CptHolzschnauz, 2021-07-25:
BIG BUG HERE:
I forgot the flags....

static unsigned long delayStarttime;
unsigned long currentTime = millis();
if (currentTime - delayStarttime >= 15000UL && signal_checked == false)
{get signal strenght;
signal_checked = true;

}
if (currentTime - delayStarttime >= 30000UL && loop_done == false)
{do a mqtt client loop
loop_done = true;}
if (currentTime - delayStarttime >= 60000UL)
{do a mqtt push, if it's the last one : delayStarttime = currentTime;
  signal_checked = false;
      loop_done = false;
}