arduino-libraries/MKRNB #80: NB_SMS struggles over the "/" sign in the date of the received sms

CptHolzschnauz, 2021-07-21:
After a sms is received, the lib ask with
AT+CMGL="REC UNREAD"
+CMGL: 0,"REC UNREAD","+TelNr",,"21 
and breaks here, the modem is freezed.

The complete output (from m-center software) is
+CMGL: 0,"REC UNREAD","+TelNr",,"21/07/21, 07:09:48+08"

So the lib strugles over the "/" sign

CptHolzschnauz, 2021-07-25:
I just double checked with another MKR1500 board. The same behaviour:

14:03:54.801 -> AT+CMGL="REC UNREAD"
14:03:54.801 -> OK
14:03:55.887 -> AT+CMGL="REC UNREAD"
14:03:55.887 -> +CMGL: 0,"REC UNREAD","+417xxxxxxxxx",,"21

There must be other out there with the same problem?

CptHolzschnauz, 2021-07-25:
It's not an issue, i have a internal problem, with another lib it works

closed 2021-07-25 by CptHolzschnauz

joshbober, 2021-09-08:
I am having this same issue where the message gets cut off, but it isn't due to the '/' character, I believe it is some kind of buffer error.
Using the official RecieveSMS sketch and a clean library install, clean compile, I have this issue every time. You say "with another lib it works", does that mean it does not work with this MKRNB library? I am submitting this as an open issue with more information from my testing.

CptHolzschnauz, 2021-09-09:
Don't trust too much the AT console. 
Do you need the whole message or just peek()?
Do the board (modem) freeze when this happens?

joshbober, 2021-09-09:
For debugging purposes, I need to see how to handle receiving SMS messages for my project. I figured the official sketch is a good starting point for that. 

The modem does not freeze, it enters the while() loop that waits for MODEM.ready() to return a non-zero number, which never happens. Please check out my post #86 where I go over more of the details. 

I just wanted to comment here to say that, while this issue was closed, the resolution was not clear. At this point I know that it isn't the '/' character that is the problem, so I wanted to share that where it's relevant. 

You mentioned using another library, can I ask which one?