arduino-libraries/MKRNB #90: AT+USOCO and AT+USOCL returning ERROR constantly when trying to connect to a server

cth000182, 2021-10-07:
It has been a couple of weeks since we found our issue, but essentially it was a one liner fix...

Background: The Arduino MKR NB 1500 was running with no issue for quite some time during testing. After about a week or two, we were unable to connect to the Firebase server anymore. We found that the sockets were not being created, opened, or closed. Then, every new MKR we tested had the same issue. After not finding solutions online, we read through the SARA-R4 datasheet and enabled Modem debug AT commands.

Fix: Long story short, in the NBClient.cpp, (line 128ish) we change `MODEM.send("AT+USECPRF=0,0,1");` to `MODEM.send("AT+USECPRF=0,0,0");` and the sockets are now functioning correctly. We have GET/POST thousands of requests with <1% failure to connect rate. 

During our troubleshooting and Googling, we realize many others have had this issue or at least, the same symptoms so we felt the need to post here for others. Hope this helps!

lizabetharum, 2024-04-05:
Thank you so much for your post. This was driving me crazy!

timkuijsten, 2025-09-03:
If I read it correctly `AT+USECPRF=0,0,0` disables validating the certificate of the remote side. This will allow a man-in-the-middle attack.