arduino-libraries/MKRNB #104: SMS receive sketch does not work

PogCarr, 2022-10-21:
The MKRNB library has a fault in it such that the example sketch for SMS Receive does not work. If there are pending messages when it boots, those messages will be shown. But subsequent messages sent will not be shown until the board is restarted.
This behavior has been confirmed with both T-mobile and Verizon networks.

PogCarr, 2022-11-19:
Following up on this...I am happy to help troubleshoot, share a screen and demo, etc. with whomever is working on this. I truly think it is an indexing issue in the code, but I am not quite good enough with C++ to debug the library code!

PogCarr, 2023-02-27:
Still hoping someone will address this Issue. To the best of my ability to find on the 'net, nobody has been able to successfully receive SMS messages sent to the MKR NB 1500 once it is on and running, regardless of what firmware, etc. is running. If someone can verifiably prove me wrong, please let me know!

Christof121, 2023-09-19:
Not a fix but a workaround.
Credits: https://forum.arduino.cc/t/mkr-nb-1500-modem-help-udp-and-sms/1153317/12
AT+CEDRXS=0,4
AT+CPSMS=0
Passing this Commands using the SeriellSARAPassthrough Example Sktech
Worked for me. Now I get the SMS after about 3 seconds without restarting.
But I haven't yet performed a long-term test