arduino-libraries/MKRNB #110: Reading a 4 byte packet from UDP server returns extra unknown characters. �&

sebastienlabine, 2023-04-06:
While trying to read a from a UDP socket, I received some unknown characters when the packet is exactly 4 byte long.

```c++
int packetSize = server.parsePacket();
if (packetSize) {
        char packet[packetSize];
        server.read(packet, packetSize);
       Serial.println(packet);
       
       // prints  open�&
}
```

However, the AT commands cleary return only "open"

+USORF: 0,"10.254.2.84",26105,4,"6F70656E"

6F70656E = open

ircole, 2023-04-07:
The input needs to be null terminated if used as a string.  One way to ensure it's null terminated is to make the character array one character larger and put a null in the last character location.  

int packetSize = server.parsePacket();
if (packetSize) {
        char packet[packetSize + 1]; // make array one character longer to allow for a null terminator
        server.read(packet, packetSize);
        packet[packetSize] = 0;  // null terminate
       Serial.println(packet);
       
       // prints  open�&
}

sebastienlabine, 2023-04-07:
@ircole Thank you for your fast answer. Is there a reason why for a packet size larger than 4 bytes it works without needing it to be null terminated?

aentinger, 2023-04-11:
Without looking deeper into that issue I'd be surprised if it would automagically work with packets larger than 4 bytes. It's probably that most of the buffer gets 0-initialised at some point and is therefore "terminated". But if you get smaller responses after larger response you'd still see remains from the previous response.

TL;DR: Always zero-terminate ;)

closed 2023-04-11 by aentinger