arduino-libraries/MKRNB #124: HELP: Unable to attach URChandler

TheGreyDiamond, 2024-02-27:
Hey folks,
I'm having issues when trying to attach a URChandler to the modem.
I'm unable to pass an argument which the compiler will accept.
This is my current code:
```cpp
<Init>
MODEM.addUrcHandler(onURCmsg);
.....
// URC handler
void onURCmsg(const String& urc) {
	// Handle URC message
}
```
This leads to `argument of type "void (ModemHelper::*)(const arduino::String &urc)" is incompatible with parameter of type "ModemUrcHandler *"`.
What am I doing, wrong here?

aentinger, 2024-02-28:
This is because `addUrcHandler` excepts a class derived from `ModemUrcHandler` and not a function pointer, i.e.

```C++
class MyModemUrcHandler : public ModemUrcHandler {
public:
  virtual void handleUrc(const String& urc) override
  {
    /* TODO: My URC handler code. */
  }
};
// ...
MyModemUrcHandler my_modem_urc_handler;
MODEM.addUrcHandler(&my_modem_urc_handler);
```

aentinger, 2024-02-28:
Note: It's an advanced feature and there's likely no need for the average user to use this. As far as I can see from the code its intended for usage by the library itself, although you can still set a hook if you should choose to do so.

closed 2024-02-28 by aentinger